rppicomidi/usb_midi_host #1: Issue enumerating Novation Launchpad Mini (1st hardware version)

denybear, 2023-08-26:
Hi,
First of all, congratulations and thank you for this brilliant piece of work. My 2 pico USB Midi host projects were previously using your own version of tinyusb, I have converted them both to your usb midi host driver!
One of my projects is using pico connected to Novation Launchpad mini (1st hardware version).

Depending on the usb cable used, enumeration goes well or gets stalled. Note that both usb cables are perfectly fine from hardware connection/cabling point of view.
Any idea where this could come from ? I have attached the traces files (for both cases of success and failure).
[fail.txt](https://github.com/rppicomidi/usb_midi_host/files/12446550/fail.txt)
[success.txt](https://github.com/rppicomidi/usb_midi_host/files/12446551/success.txt)

Thanks!

rppicomidi, 2023-08-27:
@denybear Thank you for trying my latest stuff, especially if your old stuff was working fine. I will try to help, but I need more information first because nothing pops out at me from after looking at the logs. If you you keep hardware and cable unchanged and load the old software, does it work? What version (git hash) of usb_midi_host are you using? I have fixed some issues recently. Also, in case it is a hardware issue, how is the Launchpad Mini connected to your Pico and what provides the 5V power supply?

Sadly, I don't have a launchpad mini. so duplicating your issue will be a challenge for me.

rppicomidi, 2023-08-28:
@denybear I thought about it a bit more. The point of failure in `fail.txt` is so early in the enumeration process that it hasn't even gotten enough of the USB device descriptor to know it is a MIDI device attached, so it can't be an issue with the `usb_midi_host` driver. In `fail.txt`, the host stops enumeration after it asks the device for its whole device descriptor; the control transfer never completes. That really does sound like a hardware problem such as not enough power to the Launchpad Mini, a flaky connector somewhere, an issue with your hardware board, etc. I suppose it could be an issue recently introduce to TinyUSB post revision 0.15.0, but given the problem is cable dependent, I am starting to doubt it.

Are you able to measure the VBus voltage at the Launchpad Mini?

denybear, 2023-08-28:
Hi @rppicomidi , thank you for your answer.
I encountered the same issue with the "old" software (ie. [your former implementation](https://github.com/rppicomidi/midi2usbhub)) : my project was working with some cables, but not others. So as you said, this is likely an hardware issue.
I have tried with 2 power sources so far: VBUS coming from my raspberry P400 (which I use for breadboard pico development and debugging), and my prototype works with a 5V - 500mA USB power bank (originally designed for phone charging); both sources are plugged to pico's VBUS.
As this could indeed be a powering issue, I will definitely measure VBUS voltage and try with an alternative power source. Thanks for the suggestion, I'll keep you informed asap!

denybear, 2023-08-29:
Hi @rppicomidi, I've managed to measure VBUS at the Launchpad Mini (with various power sources): 5.1V, which is normal.
Thinking about it, issue would likely come from tinyUSB. Maybe some timing issues or something like this.
My Novation Launchpad Mini works like a charm when connected to my RPI 400 under PiOs or any PC under Windows, regardless of the cable I use (including some that "don't work" when plugging the same Novation device on the pico).
I'm must say I'm a bit stucked, so any idea to pinpoint the issue is welcome. Thanks!

rppicomidi, 2023-08-31:
HI @denybear. I am not saying this is what is happening, but I have a theory. The Launchpad  Mini starts to to enumerate while it is powering up. Let's say the Vbus wire in your cable is higher resistance than your other cable, (i.e., is thinner or longer than the others). Something in the LaunchPad Mini causes a brief current surge on VBus, and that causes a brief voltage drop, and that confuses the Launchpad Mini. If you can plug a powered USB Hub into your Pico's host port and then successfully enumerate the LaunchPad Mini with your problem cable, then maybe my theory has some weight. If it does, consider the USB 2.0 spec 7.2.41. It says

_The hub downstream facing port VBUS power lines must be bypassed (CHPB) with no less than 120 µF of
low-ESR capacitance per hub. Standard bypass methods should be used to minimize inductance and
resistance between the bypass capacitors and the connectors to reduce droop. The bypass capacitors
themselves should have a low dissipation factor to allow decoupling at higher frequencies._

So maybe if you put a large capacitor to ground on the Pico VBus line (say 120 uF or larger electrolytic with 0.1 uF ceramic in parallel), it might help.

rppicomidi, 2023-09-03:
@denybear OK. I am fresh out of brilliant ideas. Sorry we couldn't solve it. I am going to close this issue. Please feel free to reopen it if you can trace where the enumeration fails. I might suggest you file a request for help in the TinyUSB project because it is the initial descriptor fetch that is failing.

rppicomidi, 2023-10-17:
@denybear I just posted what I think may be a bug fix for Issue #2. Depending on what is going on with your Launchpad, it may or
may not fix your issue. It may be worth a try. I will understand if you cannot be bothered.

denybear, 2023-10-18:
@rppicomidi Thank you for this new version. Just tested it with my Novation Launchpad (my [picopanion](https://github.com/denybear/picopanion) project), unfortunately it does not resolve my issue.
Thanks anyway for the great support !

rppicomidi, 2023-11-02:
You are welcome. I suppose it is not a surprise given that the issue occurs before it has even gotten the whole device descriptor.

closed 2023-11-02 by rppicomidi