rppicomidi/usb_midi_host #2: Working w/ MIDI devices that use Interrupt Transfers

dupontgu, 2023-10-05:
Hi - excellent work getting this together!

I've been working with a MIDI controller whose USB implementation uses interrupt transfers instead of bulk. As it stands, this library does not work with it - it enumerates fine, but no controller data flows through. I don't feel like my understanding of USB is strong enough to pitch this as an actual solution, but I've found that if you get rid of [this check](https://github.com/rppicomidi/usb_midi_host/blob/f5a17e17364c1ee5d3d56967a53a9db1e5aeee83/usb_midi_host.c#L168C1-L171C6) in the `midih_xfer_cb` function, the controller seems to work! 

From what I gather, interrupt-based endpoints need to be polled? I think the [`usbh_edpt_xfer` call](https://github.com/rppicomidi/usb_midi_host/blob/f5a17e17364c1ee5d3d56967a53a9db1e5aeee83/usb_midi_host.c#L200C16-L200C124) needs to run even if there's no data. This seems to align with what's happening in tinyusb's HID host code.

Again - I would have offered this as a PR if I was confident I weren't breaking anything else. 

I'm testing using the TouchMe controller and a Pico W running [this project](https://github.com/rppicomidi/ble-midi2usbhost/tree/main).

rppicomidi, 2023-10-16:
@dupontgu The information in the issue report does not add up to me. Would you please post a copy of your device's USB  descriptor? USB MIDI 1.0  compliant devices use bulk transfers. Also, I believe the RP2040's USB hardware would need to be set up for interrupt instead of bulk transfers.

That said, the change you suggest might be a bug find. Thank you! I will investigate.

rppicomidi, 2023-10-16:
@dupontgu the `usb_midi_host` driver does not check if the endpoint is bulk or not, and it does properly initialize the endpoint hardware for interrupts if that is what is in the USB descriptor, so this issue report is not as strange as I thought. Nevertheless, I am curious to see the interrupt endpoint in the USB descriptor because that is not supposed to be supported until USB MIDI 2.0, I thought.

dupontgu, 2023-10-16:
@rppicomidi Sure!
raw:
```
1c 00 00 00 00 00 00 00 00 00 00 00 00 00 08 00
01 01 00 03 00 80 02 65 00 00 00 03 09 02 65 00
02 01 00 80 32 09 04 00 00 00 01 01 00 00 09 24
01 00 01 09 00 01 01 09 04 01 00 02 01 03 00 00
07 24 01 00 01 41 00 06 24 02 01 01 00 06 24 02
02 02 00 09 24 03 01 03 01 02 01 00 09 24 03 02
04 01 01 01 00 09 05 01 03 08 00 0a 00 00 05 25
01 01 01 09 05 81 03 08 00 0a 00 00 05 25 01 01
03
```

And then the WireShark breakdown is attached if it's helpful.
[touchme_usb_desc_dump.txt](https://github.com/rppicomidi/usb_midi_host/files/12920858/touchme_usb_desc_dump.txt)

rppicomidi, 2023-10-17:
@dupontgu I have pushed new code to the main branch that I believe should fix your issue. Please go to the directory in your workspace that contains the `usb_midi_host` and checkout  `main` and pull the latest code. Please report back here whether the change I pushed solved your issue. If it works for you, I will update the git submodules in all of my projects that use this driver to the latest. That way, if you based your project on [ble-midi2usbhost](https://github.com/rppicomidi/ble-midi2usbhost), you can just rebase to that. Thank you again for finding this issue.

Just curious, does the TouchMe work correctly with a computer without installing a driver? Interrupt endpoints are definitely not class compliant.

dupontgu, 2023-10-17:
Just rebased and it works great! Thanks.

The TouchMe does work with my M1 Mac Mini. Also tested on Android, but _only_ when a MIDI connection is explicitly made by an app. That's what sent me down this road.

Not completely surprised it's not class-compliant - they're using a ATMega16A which does not have a USB peripheral (not that I can really judge sitting here with my PIO-based USB project 😄). I have a very strong hunch that they're using the USBMIDI Arduino library under the hood ([this line](https://github.com/BlokasLabs/USBMIDI/blob/master/src/usbmidi_vusb.cpp#L191) kinda gives it away, based on what we see here).

closed 2023-10-17 by rppicomidi