rppicomidi/usb_midi_host #6: Arduino integration

KBLiveSolutions, 2023-12-15:
Hi

Not really an issue but I was wondering how I could use your code in an arduino sketch.
I've managed to get the [Pico PIO USB device_info](https://github.com/sekigon-gonnoc/Pico-PIO-USB/blob/main/examples/arduino/device_info/device_info.ino) sketch run, now the next step is to be able to connect my MIDI devices.

rppicomidi, 2023-12-17:
@KBLiveSolutions I do not really know because I have never used Arduino. I am guessing from the example you linked that Arduino code is compatible with C code because . If that is true, look in file `usb_midi_host.h` for the callback function descriptions and the file `example/midi_host_example.c`. At a minimum, you will need to implement 
```
tuh_midi_mount_cb(uint8_t dev_addr, uint8_t in_ep, uint8_t out_ep, uint8_t num_cables_rx, uint16_t num_cables_tx)
tuh_midi_umount_cb(uint8_t dev_addr, uint8_t instance);
tuh_midi_rx_cb(uint8_t dev_addr, uint32_t num_packets);
```
If you get working Arduino code, I would welcome a pull request, but I won't be able to merge it until I get my own Arduino environment working.

rppicomidi, 2023-12-18:
@KBLiveSolutions My answer to your question could not have been helpful to you. I am sorry. I decided to try to get this to work with Arduino myself. I downloaded the Arduino IDE and got as far as you did. According to the Arduino docs, you should be able to put the library in the library folder, include it in a sketch using `#include`, and be done with it it. However, it did not work for me. I don't know enough to know why. I am going to keep digging. If you happen to make progress with this, I would appreciate it if you would keep me posted. I will do the same for you.

KBLiveSolutions, 2023-12-18:
@rppicomidi your answer was very useful and I managed to get your code working in my Arduino sketch.
However, I had to use the Adafruit_TinyUSB usb_descriptor files as those in your code were conflicting with the Arduino library. Thus, I commented out all the functions that refer to cloning in the main app.
Is it really bad? What is their role?

rppicomidi, 2023-12-18:
@KBLiveSolutions What main app are you attempting to build? What hardware are you running? If you have a device with a Pico-PIO-USB host connector and are trying to set up the RP2040 USB device interface as a MIDI interface too, then the descriptor cloning functions are required if you want the USB device interface to look like the device connected to the USB Host interface.

FWIW, I did finally get my library to build and run in an Arduino sketch using the Adafruit_TinyUSB library and Pico-PIO-USB for the host. I did not attempt to get the USB device port to do anything other than provide the console and software download for the Pico board. I will be posting the code shortly.

KBLiveSolutions, 2023-12-19:
Hey
Thanks for your time.
I quickly tried your example but it doesn't detect any MIDI device connected to my USB port.
Here is a zip containing a sketch that works for me. Note: the host pin is set to 0.
I have stripped it down to get rid of functions I didn't need in order to make the .ino page cleaner (this is part of a bigger project).

[pico-usb-midi.zip](https://github.com/rppicomidi/usb_midi_host/files/13713322/pico-usb-midi.zip)

rppicomidi, 2023-12-19:
@KBLiveSolutions Thank you for your code example. Not sure why you could not get the example to work. I suppose I will get other bug reports. Are your issues for this project resolved? If so, please close the issue.

closed 2023-12-21 by KBLiveSolutions