rppicomidi/usb_midi_host #8: Support with MIDI devices that also doubles as Audio Interface

VocasoGK, 2024-02-04:
I've tested both device_info.ino example from the Pico PIO USB Library and usb_midi_host_pio_example and both are working fine on my Zoom G1XFour. The G1XFour only has MIDI capabilities and no Audio Interface built-in. Meanwhile my NU-X MG-400 and Valeton GP-100 both has audio interface and both went undetected on those program sketches. Serial monitor only shows device removed when the device is disconnected, but nothing is happening when I connect the device. 
[image]
 Is there any workaround to this? Thanks!

rppicomidi, 2024-02-05:
There is probably no workaround right yet. Please attach a dump of the USB descriptors of the devices that are giving you trouble.

VocasoGK, 2024-02-05:
I've attached the USB descriptors dump for the Valeton GP-100 using dumper from Thesycon.

[DescriptorDump_VALETON_GP-100_PRODUCT.txt](https://github.com/rppicomidi/usb_midi_host/files/14168754/DescriptorDump_VALETON_GP-100_PRODUCT.txt)

rppicomidi, 2024-02-05:
Thanks. I will try crafting a solution based on this data, but it may take a day or two, and it may require some iteration to get it
right because I don't have your hardware. Are you willing to work with me on this?

rppicomidi, 2024-02-05:
@VocasoGK By working with me, I meant that I will put out some code, you will test it and give feedback here.

rppicomidi, 2024-02-05:
@VocasoGK This may not be a bug in this library. The default max TinyUSB Host descriptor storage is 256 bytes. I added up the lengths of all the USB Descriptors of the GP-100 and got 431 bytes. You didn't say whether you were using the library in Arduino or in C/C++. In C/C++, find your project's file `tusb_config.h` and look for a code block like this:
```
// Size of buffer to hold descriptors and other data used for enumeration
#define CFG_TUH_ENUMERATION_BUFSIZE 256
```
Change 256 to 512 and see if it works.

If you are using the Arduino Adafruit TinyUSB Host library, your project won't have a `tusb_config.h` file. Instead, that value is configured in the TinyUSB library code. Find where you are storing your Adafruit TinyUSB library version of `tusb_config.h` for the processor you are using (it is usually something like  `${ARDUINO_SKETCH_ROOT}/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/ports/${YOUR_PROCESSOR}/tusb_config_${YOUR_PROCESSOR}.h`). For example, I keep my Arduino sketches in `${HOME}/Documents` and my processor is the rp2040 that runs on a Raspberry Pi Pico board, so for me, the file is `${HOME}/Documents/Arduino/libraries/Adafruit_TinyUSB_Library/src/arduino/ports/rp2040/tusb_config_rp2040.h`. 

Please let me know whether you are using C/C++ or Arduino, what is your target hardware, and whether the change I suggested you make works for you.

rppicomidi, 2024-02-06:
For what it is worth, I submitted [this](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/375) issue to the Adafruit_TinyUSB_Arduino project.

VocasoGK, 2024-02-06:
Thanks for the quick response!
Sorry I was not being specific. I'm working on an Arduino environment, and my board is an RP2040 Zero by Waveshare.

I'll keep you updated as soon as I can.

VocasoGK, 2024-02-06:
Update:
So, the solution did make some progress. Now both device_info and usb_midi_host_pio Arduino sketches can detect and show the device's descriptor. But, the MIDI device itself is still undetected. (2nd image)

I don't have the GP-100 in my hand right now, but my MG-400 shouldn't be too different. (I've attached MG-400 Descriptor Dump just in case).

And one thing I just found out to keep in mind, I don't find any way to edit the enumeration buffer size for the EZ_USB_MIDI_HOST. It still shows just the "Device removed, address = 1" just like when the enum buffer is 256.

I really hope my explanation is clear because those kind of stuffs isn't really my forte, but I'll try as much as I can.
Thanks for the help so far!

device_info.ino dump:
[image]

usb_midi_host_pio dump: (MG-400 compared to G1XFOUR, MIDI Device Address is missing for the MG-400)
[image]

NUX MG-400 Descriptor Dump:
[DescriptorDump_NUX_MG-400.txt](https://github.com/rppicomidi/usb_midi_host/files/14182169/DescriptorDump_NUX_MG-400.txt)

rppicomidi, 2024-02-06:
@VocasoGK I can explain some. `EZ_USB_MIDI_HOST` is just another layer of software on top of `usb_midi_host`, which is an add-on to `Adafruit_TinyUSB`. So as long if you edit the file I asked you to, that should the problem of not being able to read the descriptor in the first place. The second output dump shows that the driver is successfully enumerating the device, but it looks like it is not assigning the correct endpoint addresses to the endpoints based on the USB descriptor dumps you sent me. On the MG-400, MIDI is using endpoint address IN 2 for the IN endpoint and OUT 2 for the output address. The dump says endpoint 3 & 4. The graphic screen dumps are a bit truncated on my screen. Is it possible to capture text instead?

I need a bit more information from you to help you further and determine whether this is a bug in my software, the environment,
or your hardware. Are you running one of the example sketches for `usb_midi_host`? If so, which one (please send me the exact file name)? How are you attaching the USB host port to your board? Are you using PIO for the USB host, or are you using the native USB port?

We will get to the bottom of this, I hope.

rppicomidi, 2024-02-07:
@VocasoGK I looked at your console output again. I was confused by the text before it. It looks like your ZoomG1XFour enumerated OK and presumably has the correct endpoints. The MG-400 device descriptor decoded OK, but the audio descriptors failed to parse. Am I correct? In the future, please only post console output for one device enumeration attempt at a time.

VocasoGK, 2024-02-07:
> In the future, please only post console output for one device enumeration attempt at a time.

My apologies. I've realized my mistakes.

> ZoomG1XFour enumerated OK and presumably has the correct endpoints. The MG-400 device descriptor decoded OK, but the audio descriptors failed to parse

Yes, I believe that's what happened here. 

> Are you running one of the example sketches for usb_midi_host? If so, which one (please send me the exact file name)? How are you attaching the USB host port to your board? Are you using PIO for the USB host, or are you using the native USB port?

I'm using `usb_midi_host_pio_example.ino` sketch with a modification of 
`#define HOST_PIN_DP 16 ` 
to 
`#define HOST_PIN_DP 2`

I'm using PIO to attach the USB device. I use pin 2 because pin 16 on RP2040 Zero is hardwired to the internal LED, and pin 17 is basically inaccessible.

> The graphic screen dumps are a bit truncated on my screen. Is it possible to capture text instead?

Yes sure.
This is when the RP2040 is booted up 
```
Core1 setup to run TinyUSB host with pio-usb
TinyUSB MIDI Host Example
```

then I connected the MG-400 to the RP2040:
```
Device attached, address = 1
  iManufacturer       1     NUX
  iProduct            2     NUX MG-400
  iSerialNumber       3     230605110451
```

Then when I disconnect the MG-400:
```
Device removed, address = 1
Unused MIDI device address = 1, instance = 0 is unmounted
```

And this is when G1XFour is connected:
```
MIDI device address = 1, IN endpoint 4 has 1 cables, OUT endpoint 3 has 1 cables
Device attached, address = 1
  iManufacturer       1     ZOOM Corporation
  iProduct            2     ZOOM G Series
  iSerialNumber       0     
```

and when disconnected:
```
Device removed, address = 1
MIDI device address = 1, instance = 0 is unmounted
```

> I can explain some. EZ_USB_MIDI_HOST is just another layer of software on top of usb_midi_host, which is an add-on to Adafruit_TinyUSB. So as long if you edit the file I asked you to, that should the problem of not being able to read the descriptor in the first place. 

I've just realized what's wrong, this is probably the outcome of the previous issue, where the MG-400's MIDI Device can't be mounted. My serial monitor didn't show anything because the sketch isn't really spitting out the device descriptor that are connected, it just show whenever a MIDI device is connected. TLDR: I actually fully understand what's going on with `EZ_USB_MIDI_HOST` in my case, English is hard and I can't explain, but I fully understand.

Thanks as always for the help!

rppicomidi, 2024-02-08:
@VocasoGK Thank you for the clarifications. I built a USB device from a spare Raspberry Pi Pico board that emulates the USB
descriptor you sent. As long as there is enough  `CFG_TUH_ENUMERATION_BUFSIZE` is set to 512 instead of 256, the
driver successfully parses the USB descriptor. My emulation dies shortly thereafter due to a USB stall, but before I dig any further, it would be useful to know if your system dies due to a USB stall too. Do you have the ability to attach a UART serial port to USB converter to your RP2040 Zero pins 1 and 2 (UART0)? If so, can you monitor that serial port on a separate terminal program like [putty](https://learn.adafruit.com/windows-tools-for-the-electrical-engineer/serial-terminal)? If so, you can turn on extra logging by output by editing the same `tusb_config_rp2040.h` file. Find the lines
```
#ifndef CFG_TUSB_DEBUG
#define CFG_TUSB_DEBUG 0
#endif
```
and change the 0 to a 1, 2 or 3 for different levels of debug log verbosity.

Your terminal program can capture the debug log to a file. If you attach that debug log to this bug, I might be able to figure out what is going wrong.

If you don't have the UART to USB converter, you can make one if you have a spare RP2040-Zero or Raspberry Pi Pico; just burn the Picoprobe image.

If you don't want to mess with that, I am trying to come up with a software workaround so you can get the debug log to the
Arduino serial monitor Window.

BTW, you don't have to screenshot the Arduino IDE serial monitor window. Use your mouse to highlight/select the text. Then press CTRL-C to copy it. Paste the text to a new text file in Windows notepad or some other text editor, and attach it to the bug.

Thank you for working with me on this to help me make this driver code better.

VocasoGK, 2024-02-08:
> If you don't have the UART to USB converter, you can make one if you have a spare RP2040-Zero or Raspberry Pi Pico; just burn the Picoprobe image.

Yeah I do in fact have a spare RP2040-Zero. I've tried it using Picoprobe for the past hour following the PDF manual, but I can't get anything showing up on the PuTTY. All wired properly and the ports are correct, but no data is showing up. I unfortunately don't have any USB-UART interface lying around though. 

And also, when I changed `#define CFG_TUSB_DEBUG` to `3`, that's basically just crashes the RP2040, making it detected as malfunctioned USB device on Windows. Other values than `3` making it return back to normal, but still nothing on PuTTY. Software solution via Arduino Serial Monitor might be helpful for now.

> BTW, you don't have to screenshot the Arduino IDE serial monitor window. Use your mouse to highlight/select the text. Then press CTRL-C to copy it. Paste the text to a new text file in Windows notepad or some other text editor, and attach it to the bug.

Yea, I don't know why I didn't think earlier of that.

rppicomidi, 2024-02-09:
@VocasoGK Before I start hacking code, I want to verify you went through all the steps to get the  picoprobe serial port converter working with Putty. The reason is there is less opportunity for introducing other problems this way. If you have already been through all of this, then I will send you some hacked files to try.

- Please verify with only the picoprobe and not your programmed RP2040-Zero board connected to the PC, and your Arduino IDE not running, you can see the picoprobe serial port with Putty.
- Please verify that you have wired pin GP4 from the picoprobe RP2040-Zero board to to pin GP1 of the target RP2040-Zero, Pin GP5 of the picoprobe board to pin GP0 of the target board, and that the ground pins of the two boards are connected.
- Please run a simpler sketch on your Arduino board that prints "hello, world" to UART0 (Serial1.begin(115200); Serial1.println("Hello, world");). The output should appear on your Putty terminal.

Once you have that working try again to get the debug prints working for this driver.

VocasoGK, 2024-02-09:
Oh wow, I guess technology is a weird thing.
I redid everything and now it's working properly lol.

Here's the dump with `CFG_TUSB_DEBUG` = `1`
```
MIDI opening Interface 4 (addr = 1)
MIDI opening Interface 4 (addr = 1)
MIDI descriptor parsed successfully
tu_edpt_validate 175: ASSERT FAILED
tuh_edpt_open 913: ASSERT FAILED
midih_open 479: ASSERT FAILED
```

and `CFG_TUSB_DEBUG` = `2`
[DEBUG-2.txt](https://github.com/rppicomidi/usb_midi_host/files/14216646/DEBUG-2.txt)

rppicomidi, 2024-02-09:
@VocasoGK Outstanding work! I was able to copy the configuration descriptor from the output of `DEBUG-2.txt` into a spare
Pico W I have to simulate your device and I am able to get the same assert failures. I will work on a fix.

rppicomidi, 2024-02-09:
@VocasoGK I believe I have a fix for the issue. Please download a new copy of [usb_midi_host.c](https://github.com/rppicomidi/usb_midi_host/blob/main/usb_midi_host.c) raw file and overwrite the one in the `Arduino/libraries/usb_midi_host` directory, rebuild your test application, and let me know if it works for you. If it does not work, please attach a new debug log. If it does, I will generate a new library release. Thanks again for your patience as we try to solve this issue.

VocasoGK, 2024-02-09:
Wow! It's actually working now! Doing fine on both my G1XFour and MG-400! I'll try on the GP100 probably in a couple of days, but in the meantime we can probably close this issue.

Thanks for the help for the past few days! Couldn't be more happier. One of the most helpful library I've ever used.

closed 2024-02-09 by VocasoGK

rppicomidi, 2024-02-09:
@VocasoGK You are most welcome. If you like this library, your star can encourage others to use it too.

I pushed version 1.0.2 tags to GitHub, which includes the fixes you tested and updates the README file about the enumeration buffer size. By tomorrow the changes you tested to should be available to everyone.