rppicomidi/usb_midi_host #9: Multiple devices connected?

KBLiveSolutions, 2024-02-07:
Hi

Just wondering : what would it take to be able to connect several devices using a USB hub?
Is this something that can be achieved or is there some kind of hardware limitation ?

I'm using the arduino port on a RP2040.

rppicomidi, 2024-02-08:
@KBLiveSolutions It should work with a hub now. If you look in the driver code, `_midi_host` is an array of `CFG_TUH_DEVICE_MAX` devices. The different devices are distinguished by their device addresses in the mount callback,
the unmount callback, and the receive callback. When you send MIDI to a device to transmit, you use the device address, too. So all you have to do is keep track of the mounted device addresses and track which device corresponds to which device address, and you can send MIDI among all devices in the hub. If you care to dabble in C/C++ code, I have a project called [midi2usbhub](https://github.com/rppicomidi/midi2usbhub) that provides a console UI for routing MIDI among devices, and [midi2usbhub-web](https://github.com/rppicomidi/midi2usbhub-web) that does the same thing with an embedded web browser GUI.

Did that answer your question in enough detail to get you started?

VocasoGK, 2024-02-08:
Hi there @KBLiveSolutions .
I've just tried mine with a USB hub and it works fine. I'm using the `usb_midi_host_pio_example.ino` sketch, and it assigned my devices just fine. I'm in an Arduino environment btw, and using PIO instead of USB Type-C OTG.

```
Device attached, address = 1
  iManufacturer       1     NUX
  iProduct            2     NUX MG-400
  iSerialNumber       3     230605110451
MIDI device address = 2, IN endpoint 4 has 1 cables, OUT endpoint 4 has 1 cables
Device attached, address = 2
  iManufacturer       1     Jieli Technology
  iProduct            2     SINCO
  iSerialNumber       3     5034598672735D9F
```

You can see above that my devices has its own address automatically assigned (address 1 for MG-400, and address 2 for SINCO). 

Probably one thing you should keep in mind is the power consumption/distribution, especially with bus powered devices. And probably assign your program in regards to those device addresses. Other than that I think it works pretty fine actually.

KBLiveSolutions, 2024-02-08:
Thanks both for your help.
I tried quickly yesterday and although I can connect a hub and see when devices are mounted/unmounted, I can only get MIDI data from a single device at a time (the last one connected).
Also the line ` Serial1.printf("A different USB MIDI Device is already connected.\r\nOnly one device at a time is supported in this program\r\nDevice is disabled\r\n");` made me think it was not possible.
I'll test this deeper tomorrow and let you know my results.

KBLiveSolutions, 2024-02-09:
Ok so I tried the `usb_midi_host_pio_example.ino ` sketch, and I have 2 issues:
- the `print_device_descriptor` function is blocking my `loop()`. I commented it out for now, I don't really need it.
- I can connect several MIDI devices, but I can only gat MIDI messages from the last one I plugged. Isn't it because `midi_dev_addr` is an `uint8_t` and not an array ? Or am I missing something ?

KBLiveSolutions, 2024-02-09:
Just to confirm that changing `midi_dev_addr` into an array did the trick.

rppicomidi, 2024-02-09:
@KBLiveSolutions Great! I am going to close this issue. If I did this in error, please reopen it.

closed 2024-02-09 by rppicomidi