rppicomidi/usb_midi_host #10: tuh_midi_rx_cb receiving several packets at once

KBLiveSolutions, 2024-03-05:
Hi there

I'm running `usb_midi_host.h` on a RP2040, and from time to time, `tuh_midi_rx_cb` won't work properly : I get `num_packets` values > 1, and the result is that `buffer` consists of several MIDI messages merged into one (resulting in hanging notes in the case of Note messages).
Is it a known issue ?

rppicomidi, 2024-03-06:
@KBLiveSolutions I need more information. num_packets>1 is not an issue per se. It is expected behavior if your connected USB MIDI device transmits more than one 4-byte packet in a single bulk transfer. Where you can have trouble is if the application code pulls received packets out of the receiver buffer more slowly than they come in. Please tell me
- what version of usb_midi_host is giving you trouble.
- whether your environment is C/C++ or Arduino. 
- if you are using native USB or the PIO USB for the host port.
- if you are using FreeRTOS or just a main loop
- where is the data from the USB host going? Because USB is much faster than a UART, you can certainly overrun the UART buffer buffer if you feed it too much MIDI data.

KBLiveSolutions, 2024-03-07:
Thanks for getting back to me.
I'm running the `usb_midi_host_pio_example.ino` sketch provided with the Arduino library.
I've tried it with several MIDI controllers and managed to reproduce the issue. The more data the controller sends (ie MIDI clock, aftertouch), the more errors I get.
For now the data is just printed on the Serial Monitor.
FYI, I also reproduced the issue while overclocking the Pico @240MHz.

rppicomidi, 2024-03-07:
@KBLiveSolutions
The example programs are not performance tests. They just demonstrate that the API functions. The bottleneck you are seeing happens in the function `void tuh_midi_rx_cb(uint8_t dev_addr, uint32_t num_packets)` in these lines:
```
        Serial.printf("MIDI RX Cable #%u:", cable_num);
        for (uint32_t idx = 0; idx < bytes_read; idx++) {
          Serial.printf("%02x ", buffer[idx]);
        }
        Serial.printf("\r\n");
```
`Serial.printf()` does a bunch of formatting operations and it queues data for transmission to the console, which is limited to 115.2 kbps = 11.52 kbytes/second. The Serial output buffer is finite; if data is coming in at faster than 11.52kbytes/second, which USB MIDI bulk transfers certainly are capable of, then this buffer will overflow.

Try changing the serial baud rate to 921600 (in the Serial Monitor in the `Serial.begin()` line). Try simplifying the print formatting so it prints fewer characters per message. For example,
```
        Serial.printf("%u:", cable_num);
        for (uint32_t idx = 0; idx < bytes_read; idx++) {
          Serial.printf("%02x", buffer[idx]);
        }
        Serial.printf("\r\n");
```

You can also try increasing the receiver buffer size in the driver by increasing
`CFG_TUH_MIDI_RX_BUFSIZE` to something bigger, but if your MIDI device is continually sending data faster than the application consumes it, this will only delay the problem a little while. If you want to try that, before the `include "usb_midi_host.h" line, modify the example code before the `#include ` line as follows:
```
// Add USB MIDI Host support to Adafruit_TinyUSB
#define CFG_TUH_MIDI_RX_BUFSIZE 2048
#include "usb_midi_host.h"
```
Please let me know if any of these suggestions help you.

KBLiveSolutions, 2024-03-08:
Thanks again.
I don't think it's an issue with the `Serial.print()` since I am using this code in another bigger sketch where the MIDI messages are transferred to core0 and sent over an ethernet port, and I observe the same hanging notes, even when no printing is done altogether.
Anyway, I managed to create a mechanism that separates contiguous messages and sends them separately to core0, it seems to be working fine.
I'll close this issue and create another one if the problem persists. Thanks again for your help.

KBLiveSolutions, 2024-08-30:
Hi there

I am still having issue with hanging notes resulting from packets not being processed on Arduino.
Here is my `tuh_midi_rx_cb`
`
void tuh_midi_rx_cb(uint8_t dev_addr, uint32_t num_packets)
{
        if (num_packets != 0)
        {
            uint8_t cable_num;
            uint8_t buffer[48];

            while (1) {
                uint32_t bytes_read = tuh_midi_stream_read(dev_addr, &cable_num, buffer, sizeof(buffer));
                if (bytes_read == 0)
                    return;
                  uint32_t data = createUint32FromBytes(dev_addr, buffer, 0, bytes_read);
                  write_to_other_core(data);
            }
        }
}
`

When printing the console, I can see that when num_packets>1, sometimes `tuh_midi_stream_read` will miss some packets, especially when playing clutters of notes.
I have tried to make the sketch as simple as possible, but I can't find out what the problem is.
I would greatly appreciate your help.

Here is the full .ino (note that the HOST_PIN_DP is 3)

`#include "pio_usb.h"
#include "Adafruit_TinyUSB.h"
#define HOST_PIN_DP 3  // Pin used as D+ for host, D- = D+ + 1
#define LANGUAGE_ID 0x0409
#include "usb_midi_host.h"
Adafruit_USBH_Host USBHost;
tusb_desc_device_t desc_device;
int connected_dev_amnt = 0;
uint8_t midi_dev_addr[16];

#include <MIDI.h>
Adafruit_USBD_MIDI usb_midi(2);
MIDI_CREATE_INSTANCE(Adafruit_USBD_MIDI, usb_midi, MIDI);

void core_debug(char *p_msg) {
  Serial.printf("core%d - %s", get_core_num(), p_msg);
}

bool write_to_other_core(uint32_t data) {
  if (rp2040.fifo.push_nb(data) == false) {
    core_debug((char *)"Failed to push\n");
    return false;
  }
  // core_debug((char*)"pushed\n");
  return true;
}

bool read_from_other_core(uint32_t *p_data) {
  if (rp2040.fifo.available()) {
    if (rp2040.fifo.pop_nb(p_data) == false) {
      core_debug((char *)"failed to pop\n");
      return false;
    }
    // core_debug((char*)"popped\n");
    return true;
  }
  return false;
}

//--------------------------------------------------------------------+
// String Descriptor Helper
[... 199 lines cut ...]

void loop() {
  uint32_t message;
  if (read_from_other_core(&message) == true) processCore1Data(message);
}

void setup1() {
  pio_usb_configuration_t pio_cfg = PIO_USB_DEFAULT_CONFIG;
  pio_cfg.pin_dp = HOST_PIN_DP;
  // delay(1000);
  USBHost.configure_pio_usb(1, &pio_cfg);
  USBHost.begin(1);
}

void processCore0Data(uint32_t data) {
  getConnectedDevices();
}

// core1's loop
void loop1() {
  USBHost.task();
  uint32_t message;
  if (read_from_other_core(&message) == true) processCore0Data(message);
}
`

rppicomidi, 2024-08-30:
@KBLiveSolutions I did not examine the sketch in detail, but I think there is at least one bug in the code snip in the bug report. If `num_packets` > 1, then `tuh_midi_stream_read()` will return more than 4 bytes in the buffer. If you want to pull one message at a time from the buffer because you want to encode it as a `uint32_t`, use `tuh_midi_packet_read()` instead. If you want to pull at most 4 bytes from the buffer so you can encode the stream as a `uint32_t`, make `buffer` 4 bytes or replace `sizeof(buffer)` with 4.

KBLiveSolutions, 2024-09-02:
Thanks for your help!
I managed to get rid of the hanging notes with `tuh_midi_packet_read `, so that's good news. But the reason I used `tuh_midi_stream_read ` was to be able to use packets larger than 4 bytes (Sysex).
I can live without it, but I'd prefer to have all types of messages forwarded.

My other issue now is that when forwarding the messages to the RP2040 USB port by using `usb_midi.writePacket(packet)`,  the packets sent are always 4 bytes long, resulting in Invalid messages when I have Clock messages.
Do you know any solution for that?
Should I bring it to the TinyUSB discussions?

rppicomidi, 2024-09-04:
`tuh_midi_packet_read` is reading the raw USB MIDI packet. The packet is encoded; the first byte contains the cable number (cn) 0-0xF and the code index number (cin), which, if your device encodes it properly, will describe whether the remaining 3 bytes in the packet contain a 3-byte message, a 2-byte message, a 1-byte message, or a fragment of a much longer system exclusive message. See section 4 of the [USB MIDI 1.0 Class specification](https://usb.org/sites/default/files/midi10.pdf).

If the code is passing the 4-byte array it reads from the host directly to the Arduino USB Device class method `writePacket()`, there should be no issues. If the code is encoding it first to a uint32_t and then decoding it to a packet, make sure that the decoded packet is identical to the packet received. If it is, the code should have no issues passing host data to the USB device. That applies to clock data and sysex data too. If the code is messing with the the first byte information cn/cin byte, or changing the byte order, then it will not work correctly in many cases.

I am happy to continue to try to help you, but if I am not helping, or if you want others to chime in with help, the [Adafruit_TinyUSB](https://github.com/adafruit/Adafruit_TinyUSB_Arduino) discussions may be helpful, too. Just be aware that this USB MIDI Host driver is not written or supported by Adafruit. This driver is an application level driver that uses TinyUSB USB Host support to implement the USB MIDI Class specification.

KBLiveSolutions, 2024-09-05:
once again many thanks, you nailed it ! 🙏
closing the issue now.

closed 2024-09-05 by KBLiveSolutions