rppicomidi/usb_midi_host #12: Support ESP32S3 on Arduino

rppicomidi, 2024-05-07:
**Is your feature request related to a problem? Please describe.**
@marchingband attempted to compile `usb_midi_host_example.ino`
and got the error `Compilation error: #error "Please Select USB Stack: Adafruit TinyUSB Host"`

**Describe the solution you'd like**
The example should also work with the Espressif ESP32S3 Dev Module + MAX3421E host shield. The other TinyUSB dual role examples work.

**Describe alternatives you've considered**
See discussion [here](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/360)

**Additional context**
See discussion [here](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/issues/360)

marchingband, 2024-05-08:
Just want to be clear that I am not using any shield, or additional parts, the ESP32S3 supports usb host natively.

rppicomidi, 2024-05-08:
@marchingband According to the TinyUSB website, TinyUSB does not support USB Host natively for the ESP32S3. I looked at the repository for the [espressif branch of tinyusb](https://github.com/espressif/tinyusb) and under `tinyusb/src/portable/espressif/esp32sx` and found a `dcd_esp32sx.c` for USB device but no corresponding `hcd_esp32sx.c` for the host side. The [esp-idf repo](https://github.com/espressif/esp-idf/tree/release/v5.3/components/usb) has a host controller interface, but my project is an extension of TinyUSB, not the esp-idf. This driver relies on TinyUSB support for host bulk and control endpoints and implements the MIDI spec on top of it. I can probably make this driver work with the MAX3421E host shield.

Unless you or someone else adds TinyUSB support for ESP32 host hardware, this driver is not the path to Arduino support for native USB host MIDI on ESP32. Should I close this issue?

marchingband, 2024-05-08:
Hmm, ok yah, that was my instinct, and now I think about it, that's why I opened it in the Adafruit repo initially.
I don't know a lot about tinyUSB, but I am certain it is tinyUSB (a port of it) that runs in esp-idf.
Like I said earlier, I have a working USB MIDI Host code, running right now on my esp32s3, under esp-idf.
It is not very complex, and if it's helpful, I can certainly share it with you.
I opened a feature request with esp-idf last year, and they added MIDI USB Host to their driver.

marchingband, 2024-05-08:
here is their port
https://github.com/espressif/tinyusb

marchingband, 2024-05-08:
Sorry to hear that, thanks for looking into it!

marchingband, 2024-05-08:
Would it be worth asking at tinyUSB to add host support? I know so little about usb I'm unclear if that's a reasonable request or not.

rppicomidi, 2024-05-09:
@marchingband Up to you. I will leave this issue open here. If others want me to support the MAX3421E host shield in this driver,
I will get to it. If no one responds in a few months, I will close it due to lack of interest.

BTW, I do understand that the ESP32 is a popular ecosystem, and I see the potential benefits of it for MIDI. Nevertheless, my attraction to the RP2040 and the Pico board in particular is you can implement a USB MIDI host and a USB MIDI device at the same time with only two added resistors and a USB connector on a 4 US dollar board. For 2 dollars more, you can buy a Pico W board that adds WiFi and Bluetooth support. There are enough other I/O pins to implement UARTs for UART MIDI too.

marchingband, 2024-05-09:
I certainly see the power of rp2040, although I havnt yet tried it. ESP32S3 makes a great audio target because of i2s and eMMC/SD peripherals, and the 8MB octal PSRAM available at $5, the newer P4 (out soon) has a 400mhz clock and will be even better at audio.

I am not sure if you've seen this repo, and I'm not sure if it uses tinyUSB, but I believe it works for S3 under Arduino https://github.com/touchgadget/esp32-usb-host-demos/blob/main/examples/usbhmidi/usbhmidi.ino

Maybe it helps.

In case you are curious here is my project, I am working on v2 currently.

https://www.sparkfun.com/products/21307

marchingband, 2024-05-09:
https://github.com/hathach/tinyusb/issues/2634

01GOD, 2024-08-20:
> I certainly see the power of rp2040, although I havnt yet tried it. ESP32S3 makes a great audio target because of i2s and eMMC/SD peripherals, and the 8MB octal PSRAM available at $5, the newer P4 [...]

I tried that usbhmidi.ino and it didn't seem to do anything. I flashed that on an S3 dev kit clone (with "verbose" on) and it didn't show up in DAW and didn't print anything to the serial monitor in Arduino IDE either. The TX light on the board came on and stayed on, but that was seemingly all it did. 

How to make that function?

marchingband, 2024-08-20:
@01GOD GitHub is not a good place for beginners to look for help. Please try the Arduino forum, or Reddit, or similar. This just spams the library maintainers who are very busy.

rppicomidi, 2024-08-20:
@marchingband In general, that is good advice. In my projects, I like to try to help beginners and people who may be advanced
coders but just have trouble describing the problem. Sometimes, they stumble across bugs in documentation or code that more
advanced users cannot be bothered to report.

@01GOD Please give me details on your hardware. If you are trying to get this project to work on ESP32S3, please look at this whole issue text. Right now, TinyUSB does not support the ESP32 USB host mode directly. This driver relies on TinyUSB for
hardware-level support, so I would expect nothing to work in this case.

If you are using a MAX3421E USB Host shield with your ESP32S3 board, then this is a good issue to report but it is not this bug.  Please open a new issue and carefully describe your issue using the bug template.

If your hardware is something else, please open a new issue and carefully describe your issue using the bug template.

01GOD, 2024-08-21:
> @marchingband In general, that is good advice. In my projects, I like to try to help beginners and people who may be advanced coders but just have trouble describing the problem. Sometimes, they [...]

Thanks for the considerate reply. Yesterday I tried it on a known good commercial ESP32-S3 dev kit clone board (not the S3 PCB that I designed and mentioned earlier). 

Yesterday I tried this Arduino IDE compatible code for USB MIDI from ESP32 to a laptop DAW (macOS), and it DOES function (although TinyUSB mode has to be enabled in Tools menu): 
https://github.com/esp32beans/ESP32USBMIDI/blob/main/examples/ESP32USBMIDI/ESP32USBMIDI.ino

Today I plan to hopefully make that do bi-directional MIDI (somehow adapt it to receive MIDI from a laptop DAW on ESP32-S3). 

Guidance (and code example(s) that are known to definitely function on S3) on how to accomplish MIDI from a laptop DAW to an S3 are certainly welcome. 

I assume this repo is primarily for plugging a USB MIDI controller into an MCU board and thanks for posting it and maintaining it! I met eben at the maker faire in Shenzhen (I'm not chinese, mainly prototyping there) about 7 calendar months or so ago and I am open to adding an rp2040, but I am really trying to minimize PCB (and code) complexity by using an ESP32 for all of the functionality of a cap touchscreen synth. If eben is a friend, say "Hi" for Me.

rppicomidi, 2024-08-21:
@01GOD Your response to @marchingband was unacceptable in any working technical forum. @marchingband did not insult you; your question talked about connecting a USB Host to a DAW, so assuming you are a beginner was not totally unfounded. Remember, just because you are an expert in one software area does not mean you are not a beginner in other areas. @marchingband's response was, in general, helpful to a beginner. If you post like that to any of my projects again, I will ban you.

The TinyUSB example you tried is for USB Device mode. TinyUSB supports ESP32 in USB device mode. If you need help with USB device mode MIDI on ESP32, then this is not the project for you. Please seek help in a project with a more relevant example.

If you need a MIDI USB Host mode driver on ESP32, I will repeat what I said earlier in this issue.
TinyUSB does not provide the low level host controller driver support to run the ESP32's native USB hardware in USB Host mode. This project relies on TinyUSB's host-controller driver.

If you are adding MAX3421E hardware to support USB host, TinyUSB claims to support that. I have never tried it. If you try it and it does not work, please file a new issue.

And do not post again to this issue. I am closing it.

closed 2024-08-21 by rppicomidi