rppicomidi/usb_midi_host #15: [Help Wanted] Some devices don't seem to mount

KBLiveSolutions, 2024-09-23:
Most of the MIDI controllers I've tried mount and work perfectly, but a few of them don't show up.
For instance I tried this cheap [url=https://www.amazon.fr/dp/B0CWNTH5D3?ref=ppx_yo2ov_dt_b_fed_asin_title]USB to MIDI converter[/url], and when I connect it, it doesn't light up and I don't get the "Device attached" log message.
When connecting [url=https://dualo.com/en/exquis/] this controller[/url], the lights turn on, but once again I don't have any message.
Do you have any idea about what could be the cause ?

**What is your setup like?**
- Arduino or C/C++
Arduino
- Target hardware board (Raspberry Pi Pico, Raspberry Pi Pico W, ...)
Raspberry Pi Pico
- What is your build machine (Windows PC, Linux PC, Mac)
Mac
- What build machine OS and version (Windows 11, MacOS Monterey,  Ubuntu Linux 22.04LTS, ...)
Sonoma
- Version of this library
0.6.1
- Version of build software (pico-sdk 1.5.1, Arduino IDE 2.3.0, Raspberry Pi Pico/RP2040 3.7, ...)
Arduino IDE 2.3.2
- How are you loading build images to the target (Arduino IDE+USB cable, Visual Studio Code+picoprobe, ...)
Arduino IDE+USB cable

**Attach any additional information to help explain the issue**
Attach log files, screenshots, etc.

rppicomidi, 2024-09-25:
@KBLiveSolutions I will get to your issue in a minute. However, here is a hint about putting a link in a GitHub issue. Write them like this:
```
[name of website or other descriptive text](url)
```
for example
```
[Some devices don't seem to mount](https://github.com/rppicomidi/usb_midi_host/issues/15)
```
It will look like this:
[Some devices don't seem to mount](https://github.com/rppicomidi/usb_midi_host/issues/15)

Now to address your question. There are many reasons why the device may fail to mount. The issue could be electrical or software. Generally, the main hardware issue is not providing the required voltage or power for the device. Connecting a powered hub between the Pico and the device should fix that one. The other big issue I have found is device's USB configuration descriptor is either too large or too complex for the software to parse. Sometimes the USB configuration descriptor is not USB MIDI class compliant. If you have to load a driver on a PC or Mac for your device to work with common MIDI utilities on your computer, then the device is not MIDI class compliant.

Please read the 
[TROUBLESHOOTING, CONFIGURATION, and DESIGN DETAILS](https://github.com/rppicomidi/usb_midi_host?tab=readme-ov-file#troubleshooting-configuration-and-design-details) section of the README file. Follow the web link to [this guide](https://github.com/rppicomidi/pico_usb_host_troubleshooting), too. Have a look at the USB descriptors for devices that work and compare them to USB descriptors from devices that do not work. If you cannot figure out a good reason why things are not working, attach text files that contain non-working USB descriptors. Also attach a debug log file for non-working devices.

I hope this helps.

KBLiveSolutions, 2024-09-25:
First off, thanks again for your help and patience, especially when my issues are so well documented.
Anyway, I'm having a hard time debugging this.
Changing `CFG_TUH_ENUMERATION_BUFSIZE` to 512 doesn't solve the problem.
I have tried setting `CFG_TUSB_DEBUG` to a non zero value, but this makes my Serial port disappear.
So I've added 
```
// // Debugging
int log_printf(const char * format, ...)
{
  char outstr[256];
  va_list va;
  va_start(va, format);
  int ret = vsprintf(outstr, format, va);
  // Uncomment the next line to send the debug log to the Serial1 output
  return Serial1.print(outstr);
}
```
to my sketch, and had to comment out `#define CFG_TUSB_DEBUG_PRINTF log_printf` in Adafruit_TinyUSB's tusb_config.h because otherwise it was declared twice.
Now my sketch runs fine, but I don't know see any additional debug info.
Where am I supposed to look for that?

Now on the hardware debugging itself, I've noticed another thing : when running my sketch, if I plug a *recognized* controller, I can mount it and unmount it as much as I want, but as soon as I plug my *unrecognized* controler, then the *recognized* one doesn't mount anymore, although the loop is still active, as if the host was down.
FYI, I am not using a USB hub, I connect the controlers one at a time.

rppicomidi, 2024-09-25:
@KBLiveSolutions

To make sure you are not having power or signal level issues, I do recommend connecting a _powered_ USB hub between your MIDI device and the Pico board. Only connect one MIDI device to the hub at a time. The powered hub will make sure the MIDI device is getting the right power level and it will clean up any weak signal levels between the Pico and the MIDI device. If it still doesn't work, then read further.

The Serial1 output comes out of the UART0 port on pins 1 and 2 of the Raspberry Pi Pico. You need a UART to USB converter to see that output. You need to run a separate terminal program to look at that output and to capture it to a text file. If what I just said makes no sense to you, do not worry.

You can check if the MIDI devices are USB class compliant MIDI devices. I warn you that I have very limited Mac experience, so perhaps check what I say before you do anything.

Do you have [Mac Homebrew](https://brew.sh/) installed? If not, and you are willing to do it, please install Homebrew.

If so, please install the `lsusb` command
```
brew install lsusb
```
Then plug one of the problem MIDI devices to your computer. Type
```
lsusb
```
to list the connected devices. They should list with a VID:PID hardware ID number pair. Then type
```
lsusb -d VVVV:PPPP -v
```
where `VVVV:PPPP` is the hardware ID number pair.

Copy the `lsusb` output from the terminal and save it to file. Attach it to this bug. Then do it again for your other problem devices.

Once you prove me with the full USB descriptor dump for each MIDI device that `lsusb` command provides, I can hopefully help you further.

KBLiveSolutions, 2024-09-26:
Ok thanks.
So, lsusb didn't provide enough information (not even the USB Class), so I ran ioreg instead.
Here are the results for a LaunchpadX that mounts correctly:

```
  | | +-o Launchpad X@01120000  <class IOUSBHostDevice, id 0x1000479fc, registered, matched, active, busy 0 (74 ms), retain 45>
  | |     {
  | |       "sessionID" = 2495729381821
  | |       "USBSpeed" = 1
  | |       "idProduct" = 267
  | |       "iManufacturer" = 1
  | |       "bDeviceClass" = 0
  | |       "IOPowerManagement" = {"PowerOverrideOn"=Yes,"CapabilityFlags"=32768,"MaxPowerState"=2,"DevicePowerState"=2,"DriverPowerState"=0,"ChildrenPowerState"=2,"CurrentPowerState"=2}
  | |       "bcdDevice" = 512
  | |       "bMaxPacketSize0" = 64
  | |       "iProduct" = 2
  | |       "iSerialNumber" = 3
  | |       "bNumConfigurations" = 1
  | |       "UsbDeviceSignature" = <35120b0100024158323831554239393033424544000000010100010300030101>
  | |       "USB Product Name" = "Launchpad X"
  | |       "locationID" = 17956864
  | |       "bDeviceSubClass" = 0
  | |       "bcdUSB" = 512
  | |       "kUSBSerialNumberString" = "AX281UB9903BED"
  | |       "USB Address" = 6
  | |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  | |       "kUSBCurrentConfiguration" = 1
  | |       "bDeviceProtocol" = 0
  | |       "USBPortType" = 0
  | |       "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  | |       "UsbExclusiveOwner" = "pid 2988, MIDIServer"
  | |       "USB Vendor Name" = "Focusrite - Novation"
  | |       "Device Speed" = 1
  | |       "idVendor" = 4661
  | |       "kUSBProductString" = "Launchpad X"
  | |       "USB Serial Number" = "AX281UB9903BED"
  | |       "IOGeneralInterest" = "IOCommand is not serializable"
  | |       "kUSBAddress" = 6
  | |       "kUSBVendorString" = "Focusrite - Novation"
  | |     }
```

Here is a device that lights up but doesn't mount (Exquis MPE MIDI controller) :

```

  | | +-o Exquis@01120000  <class IOUSBHostDevice, id 0x100047ac5, registered, matched, active, busy 0 (13 ms), retain 38>
  | |     {
  | |       "sessionID" = 2502100747909
  | |       "USBSpeed" = 1
  | |       "idProduct" = 7
  | |       "iManufacturer" = 1
  | |       "bDeviceClass" = 2
  | |       "IOPowerManagement" = {"PowerOverrideOn"=Yes,"DevicePowerState"=2,"CurrentPowerState"=2,"CapabilityFlags"=32768,"MaxPowerState"=2,"DriverPowerState"=0}
  | |       "bcdDevice" = 512
  | |       "bMaxPacketSize0" = 64
  | |       "iProduct" = 2
  | |       "iSerialNumber" = 3
  | |       "bNumConfigurations" = 1
  | |       "UsbDeviceSignature" = <852907000002333437373339353333303333020200010100010300>
  | |       "USB Product Name" = "Exquis"
  | |       "locationID" = 17956864
  | |       "bDeviceSubClass" = 2
  | |       "bcdUSB" = 512
  | |       "kUSBSerialNumberString" = "347739533033"
  | |       "USB Address" = 6
  | |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  | |       "kUSBCurrentConfiguration" = 1
  | |       "bDeviceProtocol" = 0
  | |       "USBPortType" = 0
  | |       "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  | |       "UsbExclusiveOwner" = "pid 2988, MIDIServer"
  | |       "USB Vendor Name" = "Intuitive Instruments"
  | |       "Device Speed" = 1
  | |       "idVendor" = 10629
  | |       "kUSBProductString" = "Exquis"
  | |       "USB Serial Number" = "347739533033"
  | |       "IOGeneralInterest" = "IOCommand is not serializable"
  | |       "kUSBAddress" = 6
  | |       "kUSBVendorString" = "Intuitive Instruments"
  | |     }
```

And here's one that doesn't light up nor mount (the cheap USB to MIDI converter):

```
  | | +-o USB2.0-MIDI@01120000  <class IOUSBHostDevice, id 0x100047af4, registered, matched, active, busy 0 (16 ms), retain 38>
  | |     {
  | |       "sessionID" = 2503003369865
  | |       "USBSpeed" = 1
  | |       "idProduct" = 29997
  | |       "iManufacturer" = 0
  | |       "bDeviceClass" = 0
  | |       "IOPowerManagement" = {"PowerOverrideOn"=Yes,"DevicePowerState"=2,"CurrentPowerState"=2,"CapabilityFlags"=32768,"MaxPowerState"=2,"DriverPowerState"=0}
  | |       "bcdDevice" = 596
  | |       "bMaxPacketSize0" = 8
  | |       "iProduct" = 2
  | |       "iSerialNumber" = 0
  | |       "bNumConfigurations" = 1
  | |       "UsbDeviceSignature" = <861a2d755402000000010100010300>
  | |       "USB Product Name" = "USB2.0-MIDI"
  | |       "locationID" = 17956864
  | |       "bDeviceSubClass" = 0
  | |       "bcdUSB" = 272
  | |       "USB Address" = 6
  | |       "kUSBCurrentConfiguration" = 1
  | |       "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBHostFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  | |       "bDeviceProtocol" = 0
  | |       "USBPortType" = 0
  | |       "IOServiceDEXTEntitlements" = (("com.apple.developer.driverkit.transport.usb"))
  | |       "Device Speed" = 1
  | |       "idVendor" = 6790
  | |       "kUSBProductString" = "USB2.0-MIDI"
  | |       "IOGeneralInterest" = "IOCommand is not serializable"
  | |       "kUSBAddress" = 6
  | |     }
```

rppicomidi, 2024-09-26:
@KBLiveSolutions Sorry, but I really need to see the real USB device descriptor and real configuration descriptor. See the instructions [here](https://github.com/rppicomidi/pico_usb_host_troubleshooting?tab=readme-ov-file#dumping-the-usb-descriptors-of-a-usb-device). The sample `lsusb` command output shows a Device Descriptor and the full Configuration Descriptor. I need to see a printout like that.

Plug the Exquis MPE MIDI controller to your Mac and type
```
lsusb -d 1a86:752d -v
```
If the output has no Device Descriptor and Configuration Descriptor headings in the output, try installing
[usbutils](https://formulae.brew.sh/formula/usbutils)
and repeat the `lsusb -d 1a86:752d -v` command above.
If that doesn't work, try, [lsusb-laniksj](https://formulae.brew.sh/formula/lsusb-laniksj)
and repeat the `lsusb -d 1a86:752d -v` command above.

I know linux-like version of the lsusb command exists for my 2012 Intel Mac Mini because I use it from time to time.

If all those fail with your Mac, do you have Linux machine (has `lsusb` built in) or a Windows PC [this app is pretty good](https://www.thesycon.de/eng/usb_descriptordumper.shtml)?

KBLiveSolutions, 2024-09-27:
ok, so here is what I got for Exquis (using the PC app). Apparently there are some errors, although it mounts correctly on my mac or on the PC.

```
Information for device Exquis (VID=0x2985 PID=0x0007):

*** ERROR: Descriptor has errors! ***

------------------------------
Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device supports USB 2.0 specification
Device address: 0x0001
Current configuration value: 0x01
Number of open pipes: 2

------------------------------
Device Descriptor:
------------------------------
0x12 bLength
0x01 bDescriptorType
0x0200 bcdUSB
0x02 bDeviceClass      (Communication Device Class)
0x02 bDeviceSubClass   (Abstract Control Model - ACM)
0x00 bDeviceProtocol  
0x40 bMaxPacketSize0   (64 bytes)
0x2985 idVendor
0x0007 idProduct
0x0200 bcdDevice
0x01 iManufacturer   "Intuitive Instruments"
0x02 iProduct        "Exquis"
0x03 iSerialNumber   "347739533033"
0x01 bNumConfigurations

-------------------------
Configuration Descriptor:
-------------------------
0x09 bLength
0x02 bDescriptorType
0x0085 wTotalLength   (133 bytes)
*** ERROR: Found error in descriptor: invalid length
0x02 bNumInterfaces
*** ERROR: Found error in descriptor: invalid length
0x01 bConfigurationValue
0x00 iConfiguration
0x80 bmAttributes   (Bus-powered Device)
0x50 bMaxPower      (160 mA)

Interface Descriptor:
------------------------------
0x09 bLength
0x04 bDescriptorType
0x00 bInterfaceNumber
0x00 bAlternateSetting
0x00 bNumEndPoints
*** ERROR: Found error in descriptor: invalid length
0x01 bInterfaceClass      (Audio Device Class)
0x01 bInterfaceSubClass   (Audio Control Interface)
0x00 bInterfaceProtocol   (Audio Protocol undefined)
0x02 iInterface   "Exquis"

AC Interface Header Descriptor:
------------------------------
0x09 bLength
0x24 bDescriptorType
0x01 bDescriptorSubtype
0x0100 bcdADC
0x0009 wTotalLength   (9 bytes)
*** ERROR: Found error in descriptor: invalid length
[... 265 lines cut ...]
          baAssocJackID( 0)       2
          baAssocJackID( 1)       3
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (Invalid)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       7
Device Status:     0x0000
  (Bus Powered)
```

hope this helps you figuring out what's going on.

rppicomidi, 2024-09-27:
@KBLiveSolutions Thank you. I will work on this as soon as I have time. Right away I see that the Exquis has a wrong DeviceClass and DeviceSubClass in the device descriptor. The descriptor parser is also reporting length errors. I will have to dig in code to see if that matters and if that is the cause of the issue.

rppicomidi, 2024-09-28:
The reason the Exquis does not work with my code is the wTotalLength field of the Configuration Descriptor is 133 (0x85). The actual length of the descriptor, if you add up the bytes, is 103 (0x67). The TinyUSB code tries to parse bytes past the end of the real configuration descriptor. There is nothing my driver, which relies on TinyUSB, can do about that. TinyUSB ignores the device class information in the Device Descriptor. Nevertheless, the bDeviceClass, bDeviceSubClass, and bDeviceProtocol fields should all be 0 for a MIDI device. If you can contact the manufacturer and request a firmware update, that should solve your problem.

Alternatively, you could request a change to the Adafruit_TinyUSB project to be more robust to incorrect wTotalLength field in the Configuration Descriptor.

Does the Exquis properly enumerate on a Windows PC without installing special software?

KBLiveSolutions, 2024-09-28:
Hi

Thanks for your feedback.
I know the devs of the Exquis, so I'll let them know. And yes, it mounts correctly on a PC without installing anything.

Did you have a moment to look at the 2nd USB device ? It seems to have errors in the bDescriptorSubtype. Similar issue ?

```

Bus 001 Device 006: ID 1a86:752d  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1a86 
  idProduct          0x752d 
  bcdDevice            2.54
  iManufacturer           0 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0075
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               96mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 [unknown]
      bInterfaceSubClass      1 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0009
        bInCollection           1
        baInterfaceNr(0)        1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         1 [unknown]
      bInterfaceSubClass      3 [unknown]
      bInterfaceProtocol      0 
      iInterface              0 
      MIDIStreaming Interface Descriptor:
        bLength                 7
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0051
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               1 Embedded
        bJackID                 2
        iJack                   0 
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               1 Embedded
        bJackID                 3
        iJack                   0 
      MIDIStreaming Interface Descriptor:
        bLength                 6
        bDescriptorType        36
        bDescriptorSubtype      2 (MIDI_IN_JACK)
        bJackType               2 External
        bJackID                 6
        iJack                   0 
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               1 Embedded
        bJackID                 7
        bNrInputPins            1
        baSourceID( 0)          6
        BaSourcePin( 0)         1
        iJack                   0 
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               2 External
        bJackID                 4
        bNrInputPins            1
        baSourceID( 0)          2
        BaSourcePin( 0)         1
        iJack                   0 
      MIDIStreaming Interface Descriptor:
        bLength                 9
        bDescriptorType        36
        bDescriptorSubtype      3 (MIDI_OUT_JACK)
        bJackType               2 External
        bJackID                 5
        bNrInputPins            1
        baSourceID( 0)          3
        BaSourcePin( 0)         1
        iJack                   0 
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 6
          bDescriptorType        37
          bDescriptorSubtype      1 (Invalid)
          bNumEmbMIDIJack         2
          baAssocJackID( 0)       2
          baAssocJackID( 1)       3
      Endpoint Descriptor:
        bLength                 9
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0020  1x 32 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (Invalid)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       7
Device Status:     0x0000
  (Bus Powered)
```

rppicomidi, 2024-09-28:
@KBLiveSolutions The second one is not obvious to me right off from the descriptor. The (invalid) tags on the bDescriptorSubtye 1 lines are not correct; see the [USB MIDI Class spec](https://usb.org/sites/default/files/midi10.pdf). The configuration descriptor length is correct at 117 bytes. I need to see the serial port debug log from the TinyUSB stack. To generate the debug log, you need to hook a UART to USB adapter to the Pico board pins 1 and 2 and you need to capture the debug output on a serial port terminal on your computer. You should capture the debug log for log level = 1 and log level = 2. If you need help doing that, please reach out to the Raspberry Pi Pico forums, the TinyUSB documentation, and the TinyUSB Discussions. Without that information, I cannot tell what is going on.

rppicomidi, 2024-11-07:
@KBLiveSolutions It has been more than a month since you filed this issue. Is this still an issue for you? Are you able to generate a debug log for the second device?

KBLiveSolutions, 2024-11-07:
I didn't have time to investigate further, the issue remains but i'll close this thread for now if you prefer.
Thanks again for your commitment.

closed 2024-11-07 by KBLiveSolutions