rppicomidi/usb_midi_host #18: Support for rp2350

ctag-fh-kiel, 2024-11-28:
Are you thinking about adding support for the rp2350? Thx :)

rppicomidi, 2024-11-29:
@ctag-fh-kiel Yes, I plan to. I think it should just work now. I don't have a Pico 2 board yet. Have you tried it and observed otherwise?

tomdude126, 2024-11-29:
I've been using this repo with my pico 2 board for 2 months and it's been working great! Specifically I'm using the built in usb port along with the midi library wrapper repo.

ctag-fh-kiel, 2024-11-29:
@tomdude126 great to hear, haven't tried myself yet...

benjaminvdb, 2024-11-29:
@tomdude126 Would you be willing to share how you got this to work? I've been struggling to get the examples to work. I managed to receive the device descriptors yesterday, but somehow failed today.

- I've tried with both the PIO approach (USB connector on a piece of stripboard with 22 Ohm resistors on the data lines) and the hardware USB connector.
- I've uploaded firmware compiled from the Arduino examples and the C-code examples, to no avail.
- What happens is that I see the "TinyUSB MIDI Host Example" message but no additional messages when I plug a USB device in (I've tried an old Korg Nanopad and an Arturia Keystep).

tomdude126, 2024-11-30:
are you sure that youve got voltage at the vbus pin? the pico needs to supply voltage to your midi device through USB since it's acting as a host. are you powering the pico through vsys instead maybe?

rppicomidi, 2024-12-01:
@benjaminvdb What version of the Pico SDK are you using? What have you tried from the [Troublshooting section of the project's README.md](https://github.com/rppicomidi/usb_midi_host/tree/main?tab=readme-ov-file#troubleshooting-configuration-and-design-details)? Have you had a look at this [troubleshooting guide](https://github.com/rppicomidi/pico_usb_host_troubleshooting)? Also, do you have a spare pico or pico 2 board? You could always program it with the standard TinyUSB [USB MIDI device example](https://github.com/hathach/tinyusb/tree/master/examples/device/midi_test) and see if that works with your pico 2 host. That will make the 5V power and ground connections simpler to measure if you do not have a breakout board. It will also give you and me a common platform to test against in case you are having issues.

rppicomidi, 2024-12-24:
@ctag-fh-kiel @benjaminvdb @tomdude126 I finally got a Pico 2 board and had some time to do some testing. I am using Pico SDK version 2.1 with the ARM core. I am using the official Raspberry Pi Pico VS Code extension. I am running my tests on a PC running Windows 11.

# Test 1 `usb_midi_host_example.c`
Used the Pico extenstion to import the project. Said no to the RISC V core. Used a simple OTG adapter on the Micro USB connector and used a USB C breakout to apply VBus 5V to Pico 2 board pin 40 and Pico 2 board pin 38 ground. Connected a Serial Port to USB adapter to pins 1 and 2 of the Pico 2 board for my serial port. I used Pico 2 board Pin 3 for ground. Built the code. I loaded the code I build using the Pico 2 board's Micro USB connector. Then I plugged the OTG adapter and a MIDI device.  It just worked.

# Test 2 `usb_midi_host_pio_example.c`
Used the Pico extenstion to import the project. Said no to the RISC V core. The project would not build because it could not find the `pio_usb.h` file. The Pico extension installed the Pico SDK in `${HOME}/.pico-sdk/sdk/2.1.0`, so I installed the Pico PIO USB library as follows:
```
cd ~/.pico-sdk/sdk/2.1.0/lib/tinyusb/hw/mcu
mkdir raspberry_pi
cd raspberry_pi
git clone https://github.com/sekigon-gonnoc/Pico-PIO-USB.git
```
I wired a USB A breakout VBus pin to Pico 2 board pin 40. I wired the breakout board GND pin to Pico 2 board pin 38. I wired the D+ pin to Pico 2 board Pin 21 and the D- pin to Pico 2 board Pin 22. I didn't bother with resistors. I connected a serial to USB converter to Pico 2 boards pins 1 and 2 and used pin 3 of the Pico 2 board for ground. I loaded the code via the Pico 2 board's Micro USB connector. It just worked.

I do not plan to test with Arduino unless someone files a bug.

My testing shows that the code just works with a RP2350 on the Pico 2 board. I am going to close this issue. @benjaminvdb If you are still having issues with this, please file a new bug as a help request. Thanks.

closed 2024-12-24 by rppicomidi

ctag-fh-kiel, 2024-12-25:
Great, thanks for verifying it. I shall be able to test as well, early next year.