rppicomidi/usb_midi_host #19: [Help Wanted] Fails to detect High-speed devices

fikryfadh, 2024-12-22:
**What help do you need?**
Hi, i've some fails to detect my ampero mini, testing with usb_midi_host.ino and ez_usb_midi_host.ino and serial monitor only stuck with this 
[image]

**What have you tried that didn't do what you expected?**
Tried change cable,resistor and usb a breakout, it doesn't seem to work out, i already test set my **#define HOST_PIN_DP**   into pin 14 and try another pin.

and edit debug value from 0 - 3 not work too

`#ifndef CFG_TUSB_DEBUG
#define CFG_TUSB_DEBUG 1
#endif`

the ampero mini is usb High-speed from what i see in system information
[image]

but im curious to test it with host shield and use  esp board its work to midi communication, is it any step that i miss to made its work? or my devices its not supported?

  
and this my usb descriptor with  % lsusb -d 84ef:0080 -v

```
Bus 020 Device 001: ID 84ef:0080  
Negotiated speed: High Speed (480Mbps)
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 [unknown]
  bDeviceSubClass         0 [unknown]
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x84ef 
  idProduct          0x0080 
  bcdDevice            1.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x019d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 Ampero Mini HS Config
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               64mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass          1 [unknown]
      bFunctionSubClass       0 [unknown]
      bFunctionProtocol      32 
      iFunction               0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 [unknown]
      bInterfaceSubClass      1 [unknown]
      bInterfaceProtocol     32 
      iInterface              5 Ampero Mini Audio
      AudioControl Interface Descriptor:
        bLength                 9
[... 396 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0100  1x 256 bytes
        bInterval               0
        bRefresh                0
        bSynchAddress           0
        MIDIStreaming Endpoint Descriptor:
          bLength                 5
          bDescriptorType        37
          bDescriptorSubtype      1 (Invalid)
          bNumEmbMIDIJack         1
          baAssocJackID( 0)       1
can't get device qualifier: Operation timed out
can't get debug descriptor: Operation timed out
Device Status:     0x0001
Self Powered
```

**What is your setup like?**
- Arduino
- Target hardware board (Waveshare RP2040-Zero )
- What is your build machine (Mac)
- What build machine OS and version (MacOS Sonoma,)
- Version of this library (usb_midi_host - 1.1.4)
- Version of build software (Version: 2.3.2)
- How are you loading build images to the target (Arduino IDE+USB cable)

rppicomidi, 2024-12-24:
@fikryfadh Sorry for the late reply. The device's configuration descriptor is 0x19d=413 bytes long. See the [README](https://github.com/rppicomidi/usb_midi_host?tab=readme-ov-file#size-of-the-enumeration-buffer). In Arduino to fix this, you will have to manually edit the `Adafruit_TinyUSB_Arduino/src/arduino/ports/rp2040/tusb_config.h` file and change the lines
```
// Size of buffer to hold descriptors and other data used for enumeration
#define CFG_TUH_ENUMERATION_BUFSIZE 256
```
to
```
// Size of buffer to hold descriptors and other data used for enumeration
#define CFG_TUH_ENUMERATION_BUFSIZE 512
```
Sadly, every time you update the library, you will have to make this manual edit again.

fikryfadh, 2024-12-24:
oke i do manually edit tusb_config.h in tinyusb library in document, with test for debug level from 0 - 3 

[image]
[image]

and in arduino using that library when upload
[image]

is it do conflict with adafruit_tinyusb library in arduino15?

it tested on usb_midi_host and ez_usb_midi_host, it still showing like this whenever i reset board and replugging ampero 
[image]
[image]

or maybe the problem is the hardware? because iam using breadboard to  wiring resistor and and usb breakout

fikryfadh, 2024-12-25:
@rppicomidi okay i edit tusb_config.h in my document libraries and arduino15 libraries to double check.

it still doesnt work out to recognize any device plug on usb a breakout, maybe the code its not the main issue for me now had to ensure the hardware, ordered new usb breakout and resistor to ensure my hardware in better connection, and order Pico W and YD-RP2040 too for backup board.

i will give any feedback later for future progress when its all arrived, and this usb A made me think can i replace into type C breakout? with 5k1 resistor for pin cc1 and cc2 to ground also 22ohm for D- and D+, just realize it can be more tiny with type C and waveshare RP2040 zero.

fikryfadh, 2024-12-27:
@rppicomidi Its work!
Big thanks for helping me, sorry taking your time for rookie mistake, the problem is my usb breakout _(broken/bad solder)_ 

[image]

And here's my tiny usb host.

rppicomidi, 2024-12-27:
@fikryfadh Fantastic! I am glad I could help. And yes, USB C instead of USB A is fine. However, A USB C host is supposed to advertise the amount of current it can provide by the appropriate pull-up resistor value on the CC1 pin. See [USB Type-C spec](https://usb.org/document-library/usb-type-cr-cable-and-connector-specification-release-24) sections 4.11 and 4.6.2.1. I am going to close this issue. Please re-open it or start a new one if you need more help.

closed 2024-12-30 by rppicomidi