rppicomidi/usb_midi_host #20: Flushing sysex data not working and/or causing latencies

Visuelle-Musik, 2025-01-28:
**Describe the bug**
When sending back several Sysex-Events to a MIDI device (Launchkey MK4) in a row only the first Sysex-Message can be sucessfully flushed (aka sent to the device), every other Sysex-Message will be put into the FiFo stack and only be flushed with the next flush-command from the application

**To Reproduce**
Steps to reproduce the behavior:
As described above: send two or more Sysex messages directly in a row and have a look how many bytes will be flushed.

**Expected behavior**
A flush should happen any time when requested by the application

**Screenshots**
init() init_step: 29
sendSysEx: sysex_sent: 10 flushed: 16  
sendSysEx: sysex_sent: 13 flushed: 0  
sendSysEx: sysex_sent: 10 flushed: 0  
sendSysEx: sysex_sent: 21 flushed: 0  
sendNoteOn, sent: 3 flushed: 0  
sendNoteOn, sent: 3 flushed: 4  
...
sendNoteOn, sent: 3 flushed: 4  
sendControlChange 104 123 1
sendCC, sent: 3 flushed: 4  
sendSysEx: sysex_sent: 10 flushed: 0  
sendSysEx: sysex_sent: 23 flushed: 0  
sendSysEx: sysex_sent: 10 flushed: 0 

**What is your setup like?**
- Arduino 
- Target hardware board (Pi Pico / RP2040, Board-Type is "Raspberry Pi Pico" in Arduino Environment)
- What is your build machine (Mac)
- What build machine OS and version: Ventura 13.6.
- Version of this library 1.1.4
- Version of build software Arduino IDE 2.3.1
- How are you loading build images to the target -> Arduino IDE+USB cable: "UF2 Board"
**Additional context**

Add any other context about the problem here.
I bypassed the problem by increasing the send-buffer to 1024 -> tuh_midih_define_limits(1024,1024,16);
but the bypass-solution leads to unwanted latencies when sending out other SysEx MIDI-data (which still may be 'stuck' in the buffer when a flush-command did not work, because the pending SysEx data has to be sent out first, is in front of the "short message" in the FiFo buffer

For flushing I use the standard method: tuh_midi_stream_flush(my_device);
I also tried to flush inside the loop() function, but this does not seem to help.
For some reason there are many situations when flushing will not work.

I had a look inside your code and I guess the critical passage is about line 770:
if(!usbh_edpt_busy(p_midi_host->dev_addr, p_midi_host->ep_out)) ...

Just as an idea: maybe you could flag not working flush-requests and send them out automatically via a background-task, when usbh_edpt is not busy anymore.

Anyways: thanks for sharing this library, it's highly apreaciated!
Mathias Brüssel

rppicomidi, 2025-01-30:
@Visuelle-Musik Thank you for filing this issue. Do you have a short program I can try to see if I can reproduce this issue in my setup?

rppicomidi, 2025-01-30:
@Visuelle-Musik The `tuh_midi_stream_flush()` causes data to be sent out over the USB host port. The RP2040 hardware has to send a complete endpoint buffer to the attached device before more data can be sent. Due to the TinyUSB HCD driver implementation and limitations of the RP2040 chip, the Pico's USB host port is relatively slow. It can take a couple of milliseconds for `tuh_midi_stream_flush()` to complete because the system only scans through the pending transfers once per millisecond. How much latency does your application experience?

You can reduce latency by packing the Host's OUT endpoint buffer up to the device's OUT endpoint buffer size with data  before flushing. Most MIDI devices have 64-byte Bulk endpoint buffers, but some have only 8 or 32 bytes. Each MIDI packet is 4 bytes, so most devices can handle 16 packets can in one buffer. In general, take the length of a sysex message in bytes, including the starting 0xF0 and 0xF7, multiply by 4, divide by 3, and round up. That will tell you how many packets you need to send the sysex message.

Your best bet is to configure the usb_midi_host driver to use larger buffers and call `tuh_midi_stream_flush()` only once in your main loop. Yes, you will have latency, but it is fine to break up sysex messages across multiple USB packets. All of the applications I have written work this way. The application does not have to flush for every write. See https://github.com/rppicomidi/usb_midi_host/tree/main?tab=readme-ov-file#config-configuration-file and the comments before https://github.com/rppicomidi/usb_midi_host/blob/829381fdf00c4563398286860c7770b6a409d06f/usb_midi_host.h#L75
for help with making buffers larger.

If I have answered your issue, please close it. Thanks.

Visuelle-Musik, 2025-01-30:
> [@Visuelle-Musik](https://github.com/Visuelle-Musik) The `tuh_midi_stream_flush()` causes data to be sent out over the USB host port. ... > If I have answered your issue, please close it. Thanks.

Hi again, thanks a lot for that reply and your suggestings. In the meantime I actually used (found out via trial&error and by looking at usb_midi_host.c) a very similar approach to what you suggest as a kind of bypass-soltution to the issue. I can't tell how much latency actually occured, but it was big enough to freeze the complete application, which was caused by the fact that the output buffer (default 64 bytes, but even when using 128 bytes) was not sufficient. I am using a buffer of 1024 just to be sure and everything works fine. Sometimes seemingly the next Note- or CC-event that get's flushed eventially will flush the Sysex in the buffer as well.
So yes, I can indeed close the issue because your answer fits very well to my own experiences. In other word I'm kind of confident that my bypass-solution actually is a fix and will not cause new issues in the future.

Still I have a suggestion for the next user that may come across the same problem: maybe you could put a short passage in the read-me hinting at the problem, maybe like "things to consider when sending out more than one sysex message in your application and when to increase your output-buffer". I read the read.me, but the behavour you explained in your answer here very good realy was not evident to me at all? Thanks again for the cool library and your answer!

Visuelle-Musik, 2025-01-30:
The issue was mainly called by a too small output buffer in combination with blocking behaviour of TinyUSB when using larger amounts of sysex-data, see reply to the issue.

closed 2025-01-30 by Visuelle-Musik