rppicomidi/usb_midi_host #21: How to add the midi driver when we use already the XInput driver ?

FreddyVRetro, 2025-01-29:
Hi,

Thanks a lot for your work, I would like to integrate in in the PicoMEM Boards.

**What help do you need?**
I use already a usb host driver for XInput and I would like to add / try your MIDI driver.

Do you know if it is possible to add more than one external driver ? How can we do this ?

**What have you tried that didn't do what you expected?**
I did not try, I don't find doc about it

**What is your setup like?**
I use Pi Pico and Pi Pico boards

rppicomidi, 2025-01-30:
@FreddyVRetro It is not really clear what you are asking. I don't know what is a PicoMEM board. Do you code in C/C++? Do you target Arduino environments? Both? Are you asking if TinyUSB supports application drivers? It is documented in TinyUSB [README](https://github.com/hathach/tinyusb?tab=readme-ov-file#host-stack).

You may add as many application drivers as you wish. See this project's [README](https://github.com/rppicomidi/usb_midi_host?tab=readme-ov-file#building-the-usb_midi_host-library-in-your-project). Basically, instead of using the implementation of `usbh_app_driver_get_cb()` in `usb_midi_host_app_driver.c`, you should make your own implementation of `usbh_app_driver_get_cb()`  that returns a pointer to the first element of an array of `usbh_class_driver_t` instead of a pointer to only one element and set `driver_count` in your function to the number of elements in that array. If you need to look at code in the host driver stack, start [here](https://github.com/hathach/tinyusb/blob/master/src/host/usbh.c#L393)

If this is what you want to know, please close this issue. If not, please try to add detail to explain what is the the problem you have. Please be sure to link to a description of your target hardware and your source code so far.

FreddyVRetro, 2025-01-30:
Hi,

PicoMEM project is here : 
https://github.com/FreddyVRetro/ISA-PicoMEM
The PicoMEM is an ISA board for old PC XT, adding RAM, Disk network (Wifi) Joystick
USB Mouse and USB Disk support are added as well.
Ultimately, I would like to add USB network but I don't find code for it :(

I use Xinput to do Joystick emulation, and would like to add MPU401 intelligent mode emulation.
As Xinput is already added via a tinyusb driver, I don't know how to add yours (The MIDI one)

FreddyVRetro, 2025-01-30:
OOPS sorry, I did not read all your answer.
So I should use an array of usb driver.
I saw the number of driver value, but did not know how to "use" it practically.

rppicomidi, 2025-01-30:
@FreddyVRetro Yes, use an array of `usbh_class_driver_t` and make `driver_count` equal to the number of elements in the array.

I assume this question is answered. I am closing the issue. This sort of question is really more about TinyUSB than the usb_midi_host driver. Further questions about application host drivers in general should be asked in the [TinyUSB discussions](https://github.com/hathach/tinyusb/discussions). Feel free to use `@rppicomidi` if you think I need my input specifically. If, in testing, you discover an issue specific to the usb_midi_host project, please do file a new issue here.

closed 2025-01-30 by rppicomidi