rppicomidi/usb_midi_host #23: note-on Events towards the MIDI-Host sometimes get lost / flush not working

Visuelle-Musik, 2025-02-07:
**Describe the bug**
This is a kind of follow-up to issue 20 https://github.com/rppicomidi/usb_midi_host/issues/20
But now it happens with note-on events or CCs too and not only sysex.

**To Reproduce**
See Issue 20, but send Note on with velocity 0 or CC with value 0 to the MIDI-Host
sendControlChange 115 127 1
sendCC, sent: 3 flushed: 4
sendControlChange 115 0 1
sendCC, sent: 3 flushed: 0

Called was: tuh_midi_stream_write(deviceId_, cableNo_, midi_cc_buffer, 3); 

Sadly there is no easy way to reproduce the issue, it happens quite rarely, but can have a big impact on the application, in the case with note-ons LEDs on the Launchkey are not updated correctly.

**Expected behavior**
Flush should always be possible, or at least be carried out delayed from the library and/or with the next flush-command from the application - seemingly the message gets completely lost.

**Screenshots**
See above

**What is your setup like?**
See https://github.com/rppicomidi/usb_midi_host/issues/20

**Additional context**
Not entierely sure if there is a connection, but I find it kind of strange, that when the condition as quoted below is happening in your code flushing is not carried out then: 
if(!usbh_edpt_busy(p_midi_host->dev_addr, p_midi_host->ep_out))

What I noticed is that directly after a MIDI-message is read from the MIDI-Host there are no issues with sening messages/flushing, but at other times it can happen that the message will not be put through at all, most likely with Sysex, but as I experienced today this can happen with CCs and note-ons as well.
So I am assuming a connection with the busy situation as quoted from your code above.

The suggestion that you gave for Sysex to fill up the MIDI-Buffer to the size of the receiver (64 bytes or similar) is an interesting approach, but tbh I don't find that practical, especially not for 3-byte messages.

Complete context beginning at line 778 of usb_midi_host.c: 
uint32_t tuh_midi_stream_flush( uint8_t dev_addr )
{
  midih_interface_t *p_midi_host = get_midi_host(dev_addr);
  TU_VERIFY(p_midi_host != NULL);

  uint32_t bytes_flushed = 0;
  if (!usbh_edpt_busy(p_midi_host->dev_addr, p_midi_host->ep_out))
  {
    bytes_flushed = write_flush(dev_addr, p_midi_host);
  }
  return bytes_flushed;
}

rppicomidi, 2025-02-09:
@Visuelle-Musik The driver returns how many bytes were flushed. It is the application responsibility to retry if flush fails. Flush fails if there is another write in progress. The code sequence write(), flush(), write(), flush() in rapid succession will likely  fail because writes take time. Your code must have a main loop or thread somewhere that calls tuh_task(). If you put a single call to flush() in that loop, and take flush away everywhere else, you should not see this issue because if flush fails the first time, the message is still in the buffer. It will be sent the next time. The example code does this.

Visuelle-Musik, 2025-02-09:
Hi, thanks for the quick answer and your advice concerning flushing repeatedly in a loop. Tbh, to me this is not very obvious in the examples. In the example you are calling send_next_note(connected); in the main loop and at the end of this procedure there is a flush. So in principle the flush takes place directly _before_ USBHost.task(); (Arduino version.) Sorry to say I don't think this is a real-world scenario, unless you only want to send out messages. Where the problems start (in my case at least) is when messages are coming in as well and have to be taken care of, like MIDI clock events for example. Apart from all that I already tried out what you are suggesting to only call flush() _once_ in the main loop, but that did not fix the problems for me as well. Also, then it can happen that sysex-messages pile up and then next note is audibly delayed, so this is not perfect either. Plus in the scenario of this issue the pending MIDI message was never sent out (or received on the Launchpad, hard to tell) at all, even not with the next MIDI message that worked. So in theory at least then the flush() from that message should have made the pending message happen, which it clearly did not. Long story short: I think it would be needed/very helpful that the library takes (at least optionally) care of such situations with a function like send_and_flush_non_blocking() ? It's really hard to tell, if not even impossible, for the application to find out when the right time for flusing is. So far I only found bypass-solutions. What I am doing now is to queue messages myself in the application and sent them out triggered by MIDI clock when nothing else is happening. Anyways, please don't get me wrong: usb_midi_host is great indeed, and I'm totally happy you provided it in the first place, but improvement on this pending messages / flushing would be great!

rppicomidi, 2025-02-10:
@Visuelle-Musik I am sorry the example code is not clear. You are correct that the example code should pull the flush call out of send_next_note(). By design, an application needs to call USBHost.task() to process an interrupts from the USB Host hardware. In the middle of the loop, the application should do all of its writes. After it finishes writing, it should then call flush(). When flush is finished, the next call to USBHost.task() processes the interrupts from completed USB transfers. The reason flush is separated from writes is speed. If you fill up the buffers with all of your writes before calling flush, you can pack more data into a single USB transfer.

I agree, I can improve the example code. In my mind, it is written differently than it is. The loop body should be
```
 USBHost.task();
send_next_note();
tuh_midi_stream_flush();
```
The `send_next_note()` function should not have flush calls in it. The [midi2usbhost](https://github.com/rppicomidi/midi2usbhost) project does things that way.

That said, you say you tried this design and it is not working for you. If your application needs to write more data than a single call to flush can flush out and it is losing transmit data, it may mean that the write buffer is not large enough. Data should not get lost in the buffer, but it is possible for writes to miss fail to write every byte. If you look at the return value of `tuh_midi_stream_write()`, is it always matching the number of bytes you wrote?

Also, I do not recommend calling flush by itself in a loop. The application needs to call USBHost.task() to properly handle interrupts from the USB OUT transfer the call to flush causes.

I will try to find some more time this week to clean things up. Thank you for your persistence.

rppicomidi, 2025-02-11:
Every USB MIDI packet has 4 bytes. Even serial port single-byte messages like MIDI Clock and two byte messages like program change consume 4 USB bytes. Sysex messages take multiples of 4 bytes. Look at section 4 of the [USB class definition for MIDI v 1.0](https://www.usb.org/sites/default/files/midi10.pdf). The first byte specifies the virtual cable number and the "Code Index Number" (CIN) which is a way to encode how many of the other 3 bytes in the packet contain useful data and, for channel messages, redundantly encodes the channel message type.

Visuelle-Musik, 2025-02-11:
Thanks for your explanation and the link to the spec! That makes sense, because of the additional data that has to be encoded. On first sight I just found it puzzling, that the number of bytes written is higher than the lenght of the buffer passed on, like here for instance: byte sent = tuh_midi_stream_write(deviceId_, cableNo_, midi_cc_buffer, 3);

rppicomidi, 2025-02-14:
@Visuelle-Musik I have pushed version 1.1.6. I imagine it will be available as an update for the Arduino IDE within 24 hours. Do the changes I made to documentation and examples suffice for closing this issue?

Visuelle-Musik, 2025-02-14:
Hi @rppicomidi thanks a lot for the new version, the enhannced examples/docs and the update to the Arduino Library. I already could install version 1.1.6 without problems and my application, as far as I can tell so far, is running exactly as before - cool!
The details in the readme about calculating bytes to be written, especially via Sysex are very helpful and hopefully will "make life easier" for other devs, too - great job!

I specifically retested the one scenario where the flush (and or sending of data to the device) did not work and sadly (without the bypass-solution of sending and flushing that perticular message twice, which is kind of 'ugly') it's still reproducable. Still have to find out if it's (only) the flush() or the messages does not get written to the device at all in the first place.

To simplify things I'll close this issue now and open another one with that specific detail, if that's ok for you?

closed 2025-02-14 by Visuelle-Musik

Visuelle-Musik, 2025-02-14:
@rppicomidi - No follow-up issue needed so far, updated the infos of the closing comment accordingly!