rppicomidi/usb_midi_host #24: Compiler warning and concern in usb_midi_host.c

robertsonics, 2025-02-18:
I'm attempting to implement this within an STM32H7 project I'm developing using Keil MDK-ARM. My project previously used the legacy STMicro USB Host stack to support a single USB MIDI Device, and I'm migrating to TinyUSB because I'd like support for a hub and multiple USB MIDI devices.

I'm getting a compiler warning in usb_midi_host.c and the line (271) does indeed look suspicious.

TU_ASSERT(usbh_edpt_xfer(dev_addr, p_midi_host->ep_out, XFER_RESULT_SUCCESS, 0));

The 3rd parameter is supposed to be a buffer pointer (uint8_t *) but is actually a macro (XFER_RESULT_SUCCESS) defined to be 0, and the warning is that the pointer is being evaluated as NULL. Is this intentional for some reason?

rppicomidi, 2025-02-26:
@robertsonics The 4th parameter is the number of bytes to transfer, which is 0, so pointer to NULL should be fine. That said, TinyUSB is in the process of merging a native USB MIDI host driver pull request. Rather than develop something new based on this driver, it might be worth your effort to try https://github.com/hathach/tinyusb/pull/1627.

robertsonics, 2025-02-28:
Will do. Thanks.

closed 2025-03-04 by robertsonics