rppicomidi/usb_midi_host #25: [Help Wanted] PIO Example project does not detect my Boss Katana

gonbabence, 2025-04-26:
**What help do you need?**
I have a [Boss Katana 100 MKII](https://www.boss.info/hu/products/katana-100_mk2/) guitar combo which I want to detect & send MIDI program changes via RPI Pico.
I tried to run the [usb_midi_host_pio_example](https://github.com/rppicomidi/usb_midi_host/blob/main/examples/arduino/usb_midi_host_pio_example/usb_midi_host_pio_example.ino) project - it detects the USB device but it looks like the `tuh_midi_mount_cb` is not called, although the `tuh_midi_umount_cb` logs the "Unused MIDI device address..." message.
Is anything I'm doing wrong? Or does the usb_midi_host lib supports devices with multiple USB interfaces? Thanks!

**What have you tried that didn't do what you expected?**
List here what you tried, what you expected to happen, and what actually happened.

**What is your setup like?**
- Arduino
- Target hardware board: **Raspberry Pi Pico**
- What is your build machine: Windows PC
- What build machine OS and version: Windows 11
- Version of this library: **1.1.6** (and Pico-PIO-USB 0.7.1)
- Version of build software: Arduino IDE 2.3.6, Raspberry Pi Pico
- How are you loading build images to the target: Arduino IDE+USB cable

**Attach any additional information to help explain the issue**

```
13:35:13.824 -> Core1 setup to run TinyUSB host with pio-usb
13:35:13.824 -> TinyUSB MIDI Host Example
13:35:16.480 -> Device attached, address = 1
13:35:16.514 ->   iManufacturer       1     BOSS
13:35:16.514 ->   iProduct            2     KATANA
13:35:16.514 ->   iSerialNumber       0     
13:35:24.836 -> Device removed, address = 1
13:35:24.836 -> Unused MIDI device address = 1, instance = 0 is unmounted
```

USB descriptor of the Katana (used USB Device Tree Viewer):
[boss_katana_100_mk2_descriptor.txt](https://github.com/user-attachments/files/19922105/boss_katana_100_mk2_descriptor.txt)

Wiring:
* DIY USB breakout board to PICO:
  - GND - GND
  - DP- - GP17
  - DP+ - GP16
  - +5V - VBUS (tried VSYS too)

rppicomidi, 2025-04-29:
@gonbabence 
I am not seeing a class compliant MIDI interface descriptor in the descriptor you sent.  The `usb_midi_host` driver only works with class compliant MIDI devices.  The USB descriptor you sent contains Vendor Specific interface class descriptors and no standard Audio Class with MIDI Streaming subclass descriptor. A class compliant MIDI device 'just works' with Windows with no special driver installed. Is it true that you need to install a driver on Windows for the Boss Katana to work on Windows? If so, then this driver will not work for you. [This program](https://www.thesycon.de/eng/usb_descriptordumper.shtml) makes a simpler USB descriptor dump output. If you don't see a MIDI Streaming subclass in the descriptor, then the `usb_midi_host` driver won't work for your application.

gonbabence, 2025-05-05:
Thanks for the quick reply!

[boss_katana_100_mk2_descriptor_tdd.txt](https://github.com/user-attachments/files/20046808/boss_katana_100_mk2_descriptor_tdd.txt)

This is the USB descriptor dump created with TDD, but no signs of MIDI Streaming subclass.

closed 2025-05-05 by gonbabence