rppicomidi/usb_midi_host #27: Example code doesn't seem to work with Pico-PIO-USB 0.7.x

HanYangZhao, 2025-05-20:
**Describe the bug**
I ran the example code with both Pico-PIO-USB 0.6.1 and Pico-PIO-USB 0.7.0 with the following outputs. The only change i made was 
`#define PIN_USB_HOST_DP  12`. I'm not using a hub. Could it be related to https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/149?

Pico-PIO-USB 0.6.1 

```
Core1 setup to run TinyUSB host with pio-usb
TinyUSB MIDI Host Example
Device removed, address = 1
Unused MIDI device address = 1, instance = 0 is unmounted
MIDI device address = 1, IN endpoint 1 has 1 cables, OUT endpoint 1 has 1 cables
Device attached, address = 1
  iManufacturer       1     Arturia
  iProduct            2     KeyStep Pro
  iSerialNumber       3     000000000001
MIDI RX Cable #0:93 2b 64 
```

Pico-PIO-USB 0.7.x

```
Core1 setup to run TinyUSB host with pio-usb
TinyUSB MIDI Host Example
```
I tried unplugging and re-plugging my midi controller but no change, it doesn't seems to get detected

**Expected behavior**
Same outputs for both versions

**What is your setup like?**
- RP2040
- usb_midi_host master branch
- arduino-pico @ 4.4.0

rppicomidi, 2025-05-20:
@HanYangZhao Your change should not matter. Do you use a USB hub? If you are not using a hub, then I do not know the answer. If you do, then I would say it is possibly is related to the the same issue. Either way, this issue needs research in the Pico-PIO-USB project or perhaps the adafruit_tinyusb_arduino project.

rppicomidi, 2025-05-22:
@HanYangZhao What version of Adafruit TinyUSB library are you using?

HanYangZhao, 2025-05-24:
I was using the bundled Adafruit TinyUSB version from arduino-pico @ 4.4.0. From the [commit  ](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/f5211f42269dc18ca7a8c6d17a8f6b04c34b644a), I think it's 3.4.1