rppicomidi/usb_midi_host #28: Adafruit TinyUSB 3.5.0 change breaks this library

maxgerhardt, 2025-05-22:
**Describe the bug**

In the future, when the library is compiled against the newly-released Adafruit TinyUSB 3.5.0 version, this library fail to compile. It uses the `USBH_EPSIZE_BULK_MAX`  enum defined in a private include header (`host/usbh_pvt.h`) 

https://github.com/rppicomidi/usb_midi_host/blob/82a5c8e04bb16d638beafbb594c8f241279c6764/usb_midi_host.c#L38-L49

Which was removed in https://github.com/adafruit/Adafruit_TinyUSB_Arduino/commit/fafc824f6b13a4e2636eda8af97fd453621070b5. All usages of this enum seem to have been replaced with `USBH_EPSIZE_BULK_MAX`, which was newly defined in the `src/host/usbh.h` header.

**To Reproduce**
Steps to reproduce the behavior:
1. See project setup in https://community.platformio.org/t/usbh-epsize-bulk-max-not-defined-in-usb-midi-host-library/47483/3?u=maxgerhardt

**Expected behavior**
Library should not include and use types from a private include header, as the things defined therein can change without notice, not being part of an official API.

A version that would for Adafruit TinyUSB below and above 3.5.0 is to redefine
```
define USED_MIDI_EPSIZE_BULK_MPS   TUD_OPT_HIGH_SPEED ? TUSB_EPSIZE_BULK_HS : TUSB_EPSIZE_BULK_FS
```
since all these values are defined in the public API `src/common/tusb_types.h.

**Screenshots**
-/-

**What is your setup like?**
- Arduino
- Target hardware board: Adafruit Feather USB Host (Raspberry Pi RP2040)
- What is your build machine: Windows
- What build machine OS and version: Windows 10
- Version of this library: 2.2.0
- Version of build software: PlatformIO, platform-raspberrypi 55cd180ac22e612e7528ddac8d3fda63e09c666d
- How are you loading build images to the target: -/-

**Additional context**

Error message when using Adafruit TinyUSB 3.5.0
```
.pio\libdeps\adafruit_feather_usb_host\usb_midi_host\usb_midi_host.c:42:35: error: 'USBH_EPSIZE_BULK_MAX' undeclared here (not in a function); did you mean 'TUSB_EPSIZE_BULK_HS'?
   42 |   #define CFG_TUH_MIDI_RX_BUFSIZE USBH_EPSIZE_BULK_MAX
      |                                   ^~~~~~~~~~~~~~~~~~~~
.pio\libdeps\adafruit_feather_usb_host\usb_midi_host\usb_midi_host.c:56:19: note: in expansion of macro 'CFG_TUH_MIDI_RX_BUFSIZE'
   56 | } midih_limits = {CFG_TUH_MIDI_RX_BUFSIZE, CFG_TUH_MIDI_TX_BUFSIZE, CFG_TUH_MAX_CABLES};
      |                   ^~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\adafruit_feather_usb_host\src\main.cpp.o
*** [.pio\build\adafruit_feather_usb_host\libf67\usb_midi_host\usb_midi_host.c.o] 
```

rppicomidi, 2025-05-22:
It sounds like you have a change that works for you. Please submit a pull request. Please note that since TinyUSB latest version has native USB MIDI Host support, this library will be deprecated soon.

tomdude126, 2025-06-04:
@rppicomidi 

"Please note that since TinyUSB latest version has native USB MIDI Host support, this library will be deprecated soon."

Where can I find more information on this? I can't seem to find where this is mentioned in the [tinyusb repo](https://github.com/hathach/tinyusb). 

Do you know of any example code that exists that utilizes using a pico as a midi usb host using only tinyusb?

rppicomidi, 2025-06-05:
@tomdude126 I have example code in C/C++; see [midi2usbhost C code version](https://github.com/rppicomidi/midi2usbhost/tree/main/C-Code) for a simple example and [midi2piousbhub](https://github.com/rppicomidi/midi2piousbhub) for a more complex C++ version. No Arduino version yet. The current configuration of the Adafruit TinyUSB library does not enable it yet, but see https://github.com/adafruit/Adafruit_TinyUSB_Arduino/tree/master/src/class/midi. Note that the TinyUSB version of the MIDI host driver code is there. It would not take much work to enable it for Arduino. I am in the middle of several other projects at the moment, but when I am done, I am going to give it a try for Arduino.

commit 75b0454b (2025-07-21): fix #28: Define USBH_EPSIZE_BULK_MAX TUH_EPSIZE_BULK_MPS
closed 2025-07-21 by rppicomidi