rppicomidi/usb_midi_host #29: C PIO example almost works

jmaltar, 2025-07-31:
First of all, thank you for this great library that, I guess, will help me a lot.

I successfully installed all the dependencies as mentioned in README.md. I can build the C PIO example where I'm using either Pico 2350 (no Risc-V version), Pico 2040 or Featherboard with USB A. With the first two development boards, I'm using a cheap USB A connector connected as VBUS - VBUS, GND - GND, D- - GP17, D+ - GP16. A MIDI controller I'm using is AKAI LPK25.

In most occasions (when it's working), things behave like this:

I'm able to connect a controller via USB and the message is:
MIDI device address = 1, IN endpoint 1 has 1 cables, OUT endpoint 1 has 1 cables

then, I have to press and release a key on the keyboard six times and after the keydown the 7th time the following message is produced:
MIDI RX: Dev 1, Cable #0:90 3d 3c 80 3d 7f 90 3d 68 80 3d 7f 90 3d 7b 80 3d 7f 90 3d 7b 80 3d 7f 90 3d 7c 80 3d 7f 90 3d 7c 80 3d 7f 90 3d 7f 80 3d 7f 

then the following messages are as expected - for each keydown I get:
MIDI RX: Dev 1, Cable #0:90 3d 6b
and for keyup:
MIDI RX: Dev 1, Cable #0:80 3d 7f 

I guess that the expected behavior is to have a print like the aforementioned every time a key is pressed and released?

Also, once a USB cable is disconnected, tuh_midi_umount_cb is not triggered. 

I installed the most recent versions of each dependency.

rppicomidi, 2025-08-04:
I cannot explain the initial behavior of 7 key presses.  I have never seen that behavior before. The Pico PIO USB library and TinyUSB have changed considerably over time, and I have not tested that combination with this code in a while. All that said, if you are programming in C, you do not need this library. USB MIDI host is now part of TinyUSB. I recommend using the native TinyUSB USB MIDI class library. The API in the TinyUSB MIDI host class library is similar to, but not the same as this library. I plan to deprecate this library for C programming soon, although I may try to put examples in place for the TinyUSB class library. If you cannot get that to work on your own, please start a discussion in the TinyUSB thread and @ me in the discussion text so I know you did it. I will attempt to help you in that context as my time permits.

jmaltar, 2025-08-04:
@rppicomidi

Thank you for you response!

1. I tried to incorporate TinyUSB MIDI Host example as you proposed - although I was able to compile the officially provided example, after a few dirty hacks, unfortunately, it was not working. I pulled the newest TinyUSB version into the 2.1.1. Pico SDK - the first dirty hack. Then, I had to declare a few variables into boards/pico.h and boards/adafruit_feather_rp2040_usb_host.h... So I stopped here.

2. I figured out, things always work when using **Pico SDK 2.0.0** except for the first 7 keypresses - so my problems are almost resolved - I can live with SDK 2.0.0

3. I was able to get rid of this "7 keypresses" problem by setting - I guess a dirty hack, CFG_TUH_MIDI_RX_BUFSIZE to 4, i.e.
```
static struct midih_limits_s {
    size_t midi_rx_buf;
    size_t midi_tx_buf;
    uint8_t max_cables;
} midih_limits = {4, CFG_TUH_MIDI_TX_BUFSIZE, CFG_TUH_MAX_CABLES};
```

tomdude126, 2025-08-09:
@rppicomidi I just saw this issue and noticed the particular issue was with the Akai LPK 25 MK, which coincidentally was one of the problematic MIDI keyboards I had posted about in my issue [here.](https://github.com/rppicomidi/EZ_USB_MIDI_HOST/issues/7). A summary of that issue was I have having issues with missed midi events when pressing multiple notes simultaneously, but I was able to fix the issue on 2 of the 3 MIDI keyboards by moving all my USB processing stuff to its own CPU core, BUT the one that I still couldn't get to work was the LPK 25 MKII. It might just be a coincidence (especially considering that I'm not using PIO version of this library, I'm using the built in USB port). But its an interesting data point regardless.

Additionally, I appreciate you posting new examples using the built in tinyusb USB midi host code , i haven't gotten around to porting my code over yet, but I will in time.

Is there an equivalent to your [[EZ_USB_MIDI_HOST](https://github.com/rppicomidi/EZ_USB_MIDI_HOST)](https://github.com/rppicomidi/EZ_USB_MIDI_HOST) library, once I port over the code to use the built in tinyusb code? I find it much easier to work with. Thanks!